FreeCAD/FreeCAD #10011: [Problem] FreeCad fault in sketcher

shawnlunny, 2023-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
I am on Windows 11 Pro 2H22 release but the version shows Windows 10 2009. This may be a bug?

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * boltsfc 2022.11.5
  * CurvedShapes 1.0.4
  * fasteners 0.4.58
  * freecad.gears 1.0.0
  * Help 1.0.3
  * QuickMeasure 2022.10.28
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

I opened a project file that I had worked on and was working fine a couple months ago. I went to pickup where I left off and now when I open the sketch and immediately close edit mode using the close button, FreeCad crashes creating a windows fault in the event viewer. To be clear the design shows no object errors or pending calculations, and the design has no degrees of freedom. Nothing was modified since this last worked so I am at a loss.

I have done just about everything you can. All drivers and apps up to date- I uninstalled and reinstalled FreeCad. I even tried running the installer as admin. I can only think that a windows change broke this but since my design is all sketched, extrusions, etc I can't use FreeCad in this state. I am unable to share the sketch as it's for a project, but I can share other logging as needed.

Faulting application name: FreeCAD.exe, version: 0.20.2.2, time stamp: 0x640cf95b
Faulting module name: TKBRep.dll, version: 7.6.3.0, time stamp: 0x62f1a45b
Exception code: 0xc0000005
Fault offset: 0x000000000001e8b3
Faulting process id: 0x0x2948
Faulting application start time: 0x0x1D9C1742A437C4E
Faulting application path: C:\Users\XXXX\AppData\Local\Programs\FreeCAD 0.20\bin\FreeCAD.exe
Faulting module path: C:\Users\XXXX\AppData\Local\Programs\FreeCAD 0.20\bin\TKBRep.dll
Report Id: 00cf8552-0e88-487c-afb1-ba14dca19873
Faulting package full name: 
Faulting package-relative application ID: 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Anthony-Gaudino, 2023-07-28:
Maybe it crashes due to recomputing.

If you try to recompute the feature that uses the sketch or the whole part does it crash?

You can also try finding the root cause by removing operations from newer to older ones, it's better to save your project as another file just in case, then go deleting newer operations and see if after that, opening and closing the sketch still crashes.
If you remove everything and leave just the sketch and opening and closing it still crashes then the problem should be in the sketch itself.
Then you can do the same in the sketch by deleting constraints and check if it still crashes.

grubertechnic, 2023-07-28:
> Faulting module name: TKBRep.dll, version: 7.6.3.0, time stamp: 0x62f1a45b

If that is true, then it have nothing to do with the sketcher.

Please retry it on the 0.21maser, there are already many bugfixes included and/or attach the file.

shawnlunny, 2023-07-29:
@Anthony-Gaudino The issue was indeed a newer operation on the part but which was not showing an error; specifically a fillet. Surprisingly I recreated it with the same exact parameters and the issue resolved.

Thanks @grubertechnic for the suggestion of grabbing a 0.21 branch but I don't see a github branch tagged with that. Only weekly development branches, but it's unclear if those are considered stable.

grubertechnic, 2023-07-29:
> Only weekly development branches, but it's unclear if those are considered stable.

Yes, that is what i mean. They are most of the time more stable as the stable version itself. Also right now.

But at least
 - attach a file
 - and step by step instruction

to reproduce the issue would be helpful (most helpful).

Also note: the last month i heard 2 times, that github have issues with downloads. Be sure, that your download is not affected, since it links to a "faulty" file. Also you can start FreeCAD in the terminal (or how it is called in windows) with `FreeCAD -l` (or modify the link) and work normal with FreeCAD. When it crash, send the logfile or the stacktrace. Then we can say maybe more.

https://wiki.freecad.org/Start_up_and_Configuration

luzpaz, 2023-07-29:
@shawnlunny https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds <--- v0.21 dev builds

shawnlunny, 2023-08-08:
After I cleared the issue I was unable to reproduce unfortunately. I will monitor and share if that changes. Thanks for the help.

closed 2023-08-08 by shawnlunny