FreeCAD/FreeCAD #10013: [Problem] TechDraw Detail View does not work for Sketches

AccidentallyTheCable, 2023-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.56
  * kicadStepUpMod 10.19.4
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

Using Detail View when the Projection View is set to a Sketch does not function, the Detail area is empty and does not update. The Sketch needs to be padded, and then that pad used for the view. Changing this, at least in my case, also really breaks all of my measurements that I had on the TechDraw page already.

Additionally, if I set the Detail View while the parent view is configured for the Pad, I will see the proper detail view. If I change the parent view to be configured for the Sketch, the Detail View stays, however does not update when moving the circle around.

### Anything else?

This limitation or requirement is not mentioned anywhere on the wiki, which is what leads me to think this is perhaps a bug, especially since the TechDraw can render sketches without problems.

Attached irc chat log from someone who helped me figure out that the sketch needed to be padded.
[irc-chat.log](https://github.com/FreeCAD/FreeCAD/files/12197909/irc-chat.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-28:
Please retry it on the 0.21master, there are already many bugfixes included and/or attach the file.

Roy-043, 2023-07-29:
Confirmed. Not entirely surprising a TD views are primarily intended for solids.

Example file:
[TD_detail_view_only_works_with_solids.zip](https://github.com/FreeCAD/FreeCAD/files/12206706/TD_detail_view_only_works_with_solids.zip)

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git)
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

AccidentallyTheCable, 2023-07-29:
@Roy-043 sorry if I misunderstand, are you saying that TD Detail View for non solids is fixed in 0.21, or that it also persists in 0.21?

grubertechnic, 2023-07-29:
persists in 0.21
[image]

luzpaz, 2023-08-16:
@grubertechnic Thanks! JFYI, please also add the full About info (so we can see the 3rd party deps you tested with).

WandererFan, 2023-08-20:
Should be fixed by commit 8caa789982.

closed 2023-08-20 by WandererFan