FreeCAD/FreeCAD #10014: [Problem] DlgGeneralImp does not follow the rest of preference page

PaddleStroke, 2023-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
0.21
```

### Subproject(s) affected?

Core

### Problem description

The general page of preference is implemented in DlgGeneralImp.cpp / ui / h. While the other pages all follow DlgSettingsXxxxxxx.cpp

It's a small thing but it makes it hard to find this page in the folder.

So I suggest renaming this DlgSettingsGeneral

Besides I think it could be best to put all the preferences files inside a gui/preferences subfolder.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2023-07-29:
> The general page of preference is implemented in DlgGeneralImp.cpp / ui / h. While the other pages all follow DlgSettingsXxxxxxx.cpp

This seems trivial, so if you make the change and do a pull request, I think we would be happy to merge it.

> Besides I think it could be best to put all the preferences files inside a gui/preferences subfolder.

@PaddleStroke Thanks for the consideration. This is subjective and probably need to be discussed on the forum where an agreement can be reached. That said, I do agree with you.

chennes, 2023-07-30:
@benj5378 I'd argue that right here (in an GitHub issue) is the "correct" place to discuss code details like directory layout. The gui folder is a bit overfull IMO, so I definitely support cleaning it up with some appropriate subfolders, including the one proposed here. In short, I agree with both suggested changes.

benj5378, 2023-07-30:
@chennes Okay, just be aware that such discussions have already taken place on the forum where there were multiple opinions.

luzpaz, 2023-08-21:
What's the status of this ?

closed 2023-08-21 by PaddleStroke