FreeCAD/FreeCAD #10018: [Problem] Notif Area should truncate messages.

PaddleStroke, 2023-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
0.21
```

### Subproject(s) affected?

Core

### Problem description

To avoid this kind of situation : 
[image]

Where the notification area show giant and totally useless errors, I think the notification area should put a max size to the error message. The full sized message being visible in the report view.

So we could truncate the string to around 1 or 2 lines, then finish by '...' if it's longer than that.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-07-30:
I agree. Is it possible to make it clickable, and show the full text in a dialog or something?

PaddleStroke, 2023-07-31:
There's already the button that opens the notification area view where we can see the past messages. In this we can elt the full message ?
By the way, I feel that this view is a bit redundant with the report view. But that's another discussion.

maxwxyz, 2024-02-27:
related: https://github.com/FreeCAD/FreeCAD/issues/10725

luzpaz, 2025-11-16:
This seems like it can be closed, right ?

PaddleStroke, 2025-11-17:
No it's still

Just made a test with a long message and it's still showing all : 

[image]