FreeCAD/FreeCAD #10020: [Problem] Flattened SVG "Edges not ordered" on part with chamfer

Chocolateandmilkwin, 2023-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Danish/Denmark (da_DK)
Installed mods: 
  * Assembly4 0.50.2
  * Estimate 0.1.2
  * WebTools
[/code]
```

### Subproject(s) affected?

Draft

### Problem description

When i use the "Flattened SVG" export option on any part with a chamfer in any place i get the error message "Edges not ordered"

Report View gets this error message

> 09:47:34  Traceback (most recent call last):
>   File "<string>", line 4, in <module>
>   File "C:\Program Files\FreeCAD 0.20\Mod\Draft\importSVG.py", line 1876, in export
>     svg.write(Draft.get_svg(ob))
>   File "C:\Program Files\FreeCAD 0.20\Mod\Draft\draftfunctions\svg.py", line 847, in get_svg
>     svg += get_path(obj, plane,
>   File "C:\Program Files\FreeCAD 0.20\Mod\Draft\draftfunctions\svgshapes.py", line 398, in get_path
>     raise ValueError('edges not ordered')
> <class 'ValueError'>: edges not ordered

I checked out the python script "svgshapes.py" and found these lines of code from the error message
```
393 if edgeindex == 0:
394     v = get_proj(vertex[0].Point, plane)
395     edata += 'M {} {} '.format(v.x, v.y)
396 else:
397     if (vertex[0].Point - previousvs[-1].Point).Length > 1e-6:
398         raise ValueError('edges not ordered')```

If i comment out line 396-398 then the problem disappears and i can export what i need, don't know if this causes problems elsewhere though.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-07-31:
Can't reproduce this. You need to attach a file.

Chocolateandmilkwin, 2023-07-31:
I made this file where i get the error
[Chamfer.zip](https://github.com/FreeCAD/FreeCAD/files/12216352/Chamfer.zip)

Roy-043, 2023-07-31:
Thanks for the file. Confirmed. The problem also occurs in V0.21.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git)
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

Syres916, 2023-08-03:
What needs to be made clear is this only applies to Part Design, if I create the same solid using Sketcher and Part Wb and export from top view, it works as expected error free.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33682 (Git)
Build type: Release
Branch: master
Hash: 9dce2525f8efe0873787f4388f10e11e168fe16b
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.3
  * A2plus
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * freecad.gears 1.0.0
  * Curves 0.6.8
  * toSketch 1.0.1
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.56
  * sheetmetal 0.2.63
```

PR #10116 (2023-08-12): Draft: Fix SVG Edges not ordered issue
Fixes #10020

See https://github.com/FreeCAD/FreeCAD/pull/10068#issuecomment-1669568201

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---
closed 2023-08-12 by Roy-043