FreeCAD/FreeCAD #10021: Why dose the face has no edges?

StromanStroman, 2023-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Chinese/China (zh_CN)
[/code]
```

### Subproject(s) affected?

None

### Problem description

[image]
I use freecad to display a stp file,and it show a face without edge,as shown above.
I expect a normal face is like the picture below.
[image]
I do not know what happened to the face,or what happened to the stp file.
Shall anyone tell me the reason why it occurs?
How can I get clear about what happened?
The strange face is created after I use BOPAlgo_Builder to split a mass of intersecting faces.It is one of the result faces.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

StromanStroman, 2023-07-31:
[facePiecesCannotTriangle9_9_0.zip](https://github.com/FreeCAD/FreeCAD/files/12212997/facePiecesCannotTriangle9_9_0.zip)
above is the wrong stp file

Roy-043, 2023-07-31:
You have not posted an STP file. But actually I think you are referring to an STL file instead. If that is the case: in the View properties of the imported mesh change the "Display Mode" to "Flat Lines" to also see the edges in the mesh.

StromanStroman, 2023-07-31:
what i posted is just stp file,not stl file.
No，I do not mean something wrong with the display.
I mean freeCAD is right,the wrong one is the stp file.
I want to know what's wrong with stp file.
What kind of error in stp file can lead to the wrong display?
| |
13207533103
|
|
***@***.***
|
---- Replied Message ----
| From | ***@***.***> |
| Date | 7/31/2023 21:49 |
| To | ***@***.***> |
| Cc | ***@***.***> ,
***@***.***> |
| Subject | Re: [FreeCAD/FreeCAD] Why dose the face has no edges? (Issue #10021) |

You have not posted an STP file. But actually I think you are referring to an STL file instead. If that is the case: in the View properties of the imported mesh change the "Display Mode" to "Flat Lines" to also see the edges in the mesh.

Roy-043, 2023-07-31:
You have posted a BREP file. I can open it in FreeCAD but it does not look like either of the images you have posted here. Please post the correct file. And also please do not answer via email. Login to GitHub and answer there.

StromanStroman, 2023-08-01:
> You have posted a BREP file. I can open it in FreeCAD but it does not look like either of the images you have posted here. Please post the correct file. And also please do not answer via email. Login to GitHub and answer there.

Oh,I am sorry,yes i uploaded a brep,not a stp file.I said that by mistake.
Brep format is the Open cascade offical format.
brep can store original cad data in contrast to stp which may change something.
So I prefer brep 
https://github.com/FreeCAD/FreeCAD/assets/13552268/fface0ea-21d1-43ed-a207-1e533f1e76d9
Please see the video above  which show abnormals

Roy-043, 2023-08-01:
The file in the video seems to match the first image in your original report. As I have already mentioned, the BREP file you have posted looks different.

So, again, please post the correct file!

StromanStroman, 2023-08-01:
> The file in the video seems to match the first image in your original report. As I have already mentioned, the BREP file you have posted looks different. And with that file I cannot reproduce the issue.
> 
> So, again, please post the correct file!
sorry but i don't quite understand what is the correct file that you want.
So would you please describe what you want me to give you?
a stp/step file,or something else?

Roy-043, 2023-08-01:
The issue cannot be reproduces with the file you have supplied ([facePiecesCannotTriangle9_9_0.zip](https://github.com/FreeCAD/FreeCAD/files/12212997/facePiecesCannotTriangle9_9_0.zip)). See the attached image.
[image]

So you need to supply a file that actually shows the problem, either as brep or as FCStd.

StromanStroman, 2023-08-01:
> The file in the video seems to match the first image in your original report. As I have already mentioned, the BREP file you have posted looks different. And with that file I cannot reproduce the issue.
> 
> So, again, please post the correct file!

[facePiecesLessThanTwo10_2_0.zip](https://github.com/FreeCAD/FreeCAD/files/12226380/facePiecesLessThanTwo10_2_0.zip)
I uploaded a stp file.
Maybe that is what you want.

StromanStroman, 2023-08-01:
> 

> The issue cannot be reproduces with the file you have supplied ([facePiecesCannotTriangle9_9_0.zip](https://github.com/FreeCAD/FreeCAD/files/12212997/facePiecesCannotTriangle9_9_0.zip)). See the [...]

could you show me a video about what you have done and how did you do that?

StromanStroman, 2023-08-01:
> The issue cannot be reproduces with the file you have supplied ([facePiecesCannotTriangle9_9_0.zip](https://github.com/FreeCAD/FreeCAD/files/12212997/facePiecesCannotTriangle9_9_0.zip)). See the [...]

Just the file i gave to you display a face with no edges in my local freecad.
Maybe freecads of you and me are different?

Roy-043, 2023-08-01:
I have tested with V0.20 and V0.21.

I have used the [Std_Import](https://wiki.freecad.org/Std_Import) command. Nothing more.

I suggest you check your preferences. Notably: [Line Color](https://wiki.freecad.org/PartDesign_Preferences#Shape_appearance). You can also rename your user.cfg file, restart FreeCAD and try again.

Version 0.20 I have used for testing:
```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

StromanStroman, 2023-08-01:
> 
[download url](https://github.com/FreeCAD/FreeCAD/releases/download/0.20.2/FreeCAD-0.20.2-WIN-x64-installer-3.exe)
this is the version freecad that i am using,you can try it.

StromanStroman, 2023-08-01:
> I have tested with V0.20 and V0.21.
>
> I have used the [Std_Import](https://wiki.freecad.org/Std_Import) command. Nothing more.
>
> I suggest you check your preferences. Notably: [Line [...]

[download url](https://github.com/FreeCAD/FreeCAD/releases/download/0.20.2/FreeCAD-0.20.2-WIN-x64-installer-3.exe)
this is the version freecad that i am using,you can try it.

StromanStroman, 2023-08-01:
> I have tested with V0.20 and V0.21.
>
> I have used the [Std_Import](https://wiki.freecad.org/Std_Import) command. Nothing more.
>
> I suggest you check your preferences. Notably: [Line [...]

All my configuration is default.

StromanStroman, 2023-08-01:
> I have tested with V0.20 and V0.21.
>
> I have used the [Std_Import](https://wiki.freecad.org/Std_Import) command. Nothing more.
>
> I suggest you check your preferences. Notably: [Line [...]

my operating system is windows11

Roy-043, 2023-08-01:
Steps for you to undertake:

Test with the current dev version:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Provide the correct brep file and the FCStd file after importing that brep file.

Provide a step-by-step scenario of what you have done.

StromanStroman, 2023-08-02:
> Steps for you to undertake:
> 
> Test with the current dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
> 
> Provide the correct brep file and the FCStd file after importing that brep file.
> 
> Provide a step-by-step scenario of what you have done.

https://github.com/FreeCAD/FreeCAD/assets/13552268/d78eed95-25bc-49cf-89a8-597e4f5abf81
I use the version you specified to display the wrong face,and it shows the same as before
Above is the video.

StromanStroman, 2023-08-02:
[Downloads.zip](https://github.com/FreeCAD/FreeCAD/files/12235651/Downloads.zip)
Here are exported files from freecad.
What I have done is drag the brep file into freecad.

Roy-043, 2023-08-02:
OK I am able to reproduce. The requirement is to start FreeCAD without a user.cfg.

Two workarounds:
1. Recompute ([Std_Refresh](https://wiki.freecad.org/Std_Refresh)) the document.
2. Call [Std_DlgPreferences](https://wiki.freecad.org/Std_DlgPreferences), Press the Apply button and then the OK button.

Roy-043, 2023-08-02:
Closing this in favor of #10042.

closed 2023-08-07 by luzpaz