FreeCAD/FreeCAD #10022: [Problem] Path PocketShape not milling last level due to "Discard empty section"

fwillo, 2023-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * freecad.gears 1.0.0
  * Manipulator 1.4.9
[/code]
```

### Subproject(s) affected?

Path

### Problem description

I'm having a body of which I want to create a PocketShape. Pocket shape is -10.5mm deep and my tool goes down in 2mm steps. The levels -2mm, -4mm, -6mm, -8mm and -10mm are covered. -10.5mm, however, is not generated with a warning message "Discard empty section". The same shape with the same operation is working in 0.19.4. Minimal example attached to this issue.

### Anything else?

[PocketShape-DiscardEmptySection.zip](https://github.com/FreeCAD/FreeCAD/files/12213881/PocketShape-DiscardEmptySection.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fwillo, 2023-08-04:
I can confirm this problem to persist in the current 0.21 Release.

IdahoEv, 2024-01-31:
I am getting a version of this error in 0.21.2 in some circumstances.  I have attached a file with two identical shapes where one reproduces the problem and the other does not.  In this file, it seems to only happen 1) On the second copy of the shape  and 2) when extensions are turned on for the pocket. 

### My version info:
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly3 0.12.2
  * Assembly4 0.50.6
  * fasteners 0.4.63
  * FeedsAndSpeeds 0.5.0
  * Manipulator 1.5.7
  * woodworking 0.21.2.33771
[/code]

### Attachments 

[image]
[Test Rabbets 04.zip](https://github.com/FreeCAD/FreeCAD/files/14118123/Test.Rabbets.04.zip)

IdahoEv, 2024-02-06:
Update: I have now seen this in multiple other circumstances not involving extensions.  Including at least two cases where the layer that failed to generate paths was the only layer in the pocket (where that layer was smaller than the maximum step-size of the milling bit).   Can provide more reproduction cases if that's useful. 

It seems very sensitive to other parameters.  I can often work around it by changing pocket pattern, between Offset and ZigZagOffset, for example -- the bug may occur with one but not the other, or vice versa.  Sometimes changing other parameters will alleviate an instance of the problem as well.

halzy, 2025-05-07:
I've just experienced this with version 1.0.

[image]

tarman3, 2025-05-07:
Please try last weekly build
https://github.com/FreeCAD/FreeCAD/tags

halzy, 2025-05-07:
Last night I tried the build from last week and the week before. After fixing the numpy loading issues in each build I was not able to load the project. If you want a copy of the project I'm happy to attach it, let me know.

opensourcecnc, 2025-10-06:
It has something to do with the preferences about the metric/imperial system and the number of decimals, combined with the geometry tolerance. I encountered this issue a while ago and, at that time, I managed to figure out what was causing it. Unfortunately, I wasn't able to replicate it now. As soon as I'll do it, I'll get back. Meanwhile, maybe my hints will lead to a solution. But I don't think it's a bug, rather a wrong configuration