FreeCAD/FreeCAD #10027: [Problem] Greyed out (disabled) sketcher tools

FEA-eng, 2023-08-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git)
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Problem description

Almost all sketcher tools (geometry and constraint tools) can become greyed out (and thus deactivated) after a sequence of clicks. The exact repeatable sequence leading to this hasn't been identified yet (the issue seems to occur somewhat randomly) but so far the most reliable way is to draw an arbitrary circle and repeat the following steps:
1. Drag it
2. Click on the background next to it

Sometimes it happens after doing it once or twice, sometimes after doing it 3 times or more and sometimes not at all. But it usually happens. On 3 different computers with Windows (10 and 11). To bring the tools back to the active state one has to click on the geometry of the sketch (clicking on the background doesn't help).

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-01:
@xtemp09 Finally someone can reproduce it, thanks for checking. It's also good to know that it's not specific to Windows. It seems to be somewhat random.

benj5378, 2023-08-05:
This traceback was quite difficult to get, found a method at last.

Eventually, we can see, that `Gui::Command::testActive` deactives the commands in Sketcher by `Action::setEnabled`.

But according to the documentation, the function isn't even meant to _change_ the commands state, it's only meant to to _test_  the state of the command.

```cpp
/// Get somtile called to check the state of the command
void testActive();
```

Though if we look at the exact line in `Gui::Command::testActive` that changes the state, it seems like the state change is very intentionally.

bool bActive = isActive();  // Command::isActive
_pcAction->setEnabled(bActive);  // Action::setEnabled

The enabled state is eventually set to the value of `isActive()` defined in `CmdSketcherCreatePoint::isActive`:

```cpp
bool CmdSketcherCreatePoint::isActive()
{
    return isCommandActive(getActiveGuiDocument());  // SketcherGui::isCommandActive
}
```

`CmdSketcherCreatePoint::isActive` just returns the value of `SketcherGui::isCommandActive`, that only return true or false depending on other functions return boolean value. 

```cpp
bool SketcherGui::isCommandActive(Gui::Document* doc, bool actsOnSelection)
{
    if (isSketchInEdit(doc)) {
        auto mode =
            static_cast<SketcherGui::ViewProviderSketch*>(doc->getInEdit())->getSketchMode();

        if (mode == ViewProviderSketch::STATUS_NONE
            || mode == ViewProviderSketch::STATUS_SKETCH_UseHandler) {

            if (!actsOnSelection) {
                return true;
            }
            else if (Gui::Selection().countObjectsOfType(Sketcher::SketchObject::getClassTypeId())
                     > 0) {
                return true;
            }
        }
    }

    return false;
}
```

Here we know that at least `mode == ViewProviderSketch::STATUS_NONE` and `mode == ViewProviderSketch::STATUS_SKETCH_UseHandler` both return false and thereby eventually causing all the Sketcher buttons to be disabled.

What I do notice though is that `SketcherGui::isCommandActive` was changed heavily in 816fa3fe3e24fcc931336a055db3ea849ca6f99a a810fa68b43181426a6a24d74a5829f172a27bc1 bae8050fac84684e915832fcfaec5e0b2e834b0b... might these changes have introduced the bug? @FEA-eng how long has this been a bug?

Well, further research show that
```cpp
mode == ViewProviderSketch::SketchStatus::STATUS_SKETCH_UseRubberBand
mode == ViewProviderSketch::SketchStatus::STATUS_SKETCH_StartRubberBand
mode == ViewProviderSketch::SketchStatus::STATUS_SELECT_Edge
mode == ViewProviderSketch::SketchStatus::STATUS_SELECT_Cross
mode == ViewProviderSketch::SketchStatus::STATUS_SKETCH_DragCurve
```

causes the commands to be disabled. Sometimes the commands get disabled and auto recovers without user interaction. Mode is set in ViewProviderSketch

So now two questions arise. Why is the SketchMode thinking we are using a rubberband? Why does the commands only auto recover some times?

I'm begging to get hungry and it's dinner time. So I'll stop for here and let someone else take over. I'd pinpoint the issue to being the SketchMode, that Sketcher falsely believes one is doing a rubberband selection. On the surface, it's seams like an easy fix, as it's just value comparison of click distances, but I'll avoid digging into it, as I know everything I touch just turns into a deep black rabbit hole. Also, ViewProviderSketch really needs some refactoring as code organization is currently non-existing where everything is just the long switch-cases.

Alright, dinner time.

```cpp
if (done
    && SbVec2f(cursorPos - DoubleClick::prvClickPos).length() < dblClickRadius
    && (SbTime::getTimeOfDay() - DoubleClick::prvClickTime).getValue() < dci) {
        // [...]
}
else {
    [...]
    if (!done) {
        Mode = STATUS_SKETCH_StartRubberBand;
        raise(SIGINT);
    }
}
```

Traceback:

```
[...]
#4  0x00007ffff6961202 in Gui::Action::setEnabled(bool)
    (this=0x555558c2c540, enable=false)
    at /home/bensay/Development/FreeCAD/src/Gui/Action.cpp:167
#5  0x00007ffff6963a27 in Gui::ActionGroup::setEnabled(bool)
    (this=0x555558c2c540, check=false)
    at /home/bensay/Development/FreeCAD/src/Gui/Action.cpp:512
#6  0x00007ffff697ac24 in Gui::Command::testActive() (this=0x555558f061d0)
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:539
        pcAction = 0x555558c2c540
        bActive = false
#7  0x00007ffff6984871 in Gui::CommandManager::testActive() (this=0x555555b18a30)
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:1924
        It = {first = "Sketcher_CompCreateCircle", second = 0x555558f061d0}
#8  0x00007ffff6fa6e7f in Gui::MainWindow::_updateActions() (this=0x7fffffffd120)
    at /home/bensay/Development/FreeCAD/src/Gui/MainWindow.cpp:1520
#9  0x00007ffff6b1b9d2 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::MainWindow::*)()>::call(void (Gui::MainWindow::*)(), Gui::MainWindow*, void**)
    (f=(void (Gui::MainWindow::*)(class Gui::MainWindow * const)) 0x7ffff6fa6cd8 <Gui::MainWindow::_updateActions()>, o=0x7fffffffd120, arg=0x7fffffffc970)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152
#10 0x00007ffff6b1b908 in QtPrivate::FunctionPointer<void (Gui::MainWindow::*)()>::call<QtPrivate::List<>, void>(void (Gui::MainWindow::*)(), Gui::MainWindow*, void**)
    (f=(void (Gui::MainWindow::*)(class Gui::MainWindow * const)) 0x7ffff6fa6cd8 <Gui::MainWindow::_updateActions()>, o=0x7fffffffd120, arg=0x7fffffffc970)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185
#11 0x00007ffff6b1b779 in QtPrivate::QSlotObject<void (Gui::MainWindow::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*)
    (which=1, this_=0x55555609ed20, r=0x7fffffffd120, a=0x7fffffffc970, ret=0x0)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418
[...]
```

benj5378, 2023-08-05:
@abdullahtahiriyo

FEA-eng, 2023-08-07:
@benj5378 Thank you for investigating this. I noticed this problem in June, but it's possible that it had been there for some time before.

luzpaz, 2023-10-11:
bumping for progress

FEA-eng, 2024-04-03:
@PaddleStroke Since Abdullah is gone, maybe you could take a look at this small but annoying issue when you have some time. The comment by @benj5378 provides a really comprehensive insight.

PaddleStroke, 2024-04-04:
I can reproduce. I'll investigate later

FEA-eng, 2024-10-24:
> I can reproduce. I'll investigate later

This issue is still relevant. Any chance you could have a look at it for 1.0?

FEA-eng, 2025-01-23:
Still relevant (and quite annoying) issue

commit 4bdaa018 (2025-07-19): Sketcher: Fix random gray icon (#22502)

* Sketcher Fix #10027
closed 2025-07-19 by kadet1090