FreeCAD/FreeCAD #10034: Sketcher: Driven constraint crashes FreeCAD

JohnOCFII, 2023-08-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33669 (Git)
Build type: Release
Branch: (HEAD detached at 0.21rc1a)
Hash: e5a00e153d36b0d53ebd568f4f47e95fbb5d0fcd
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Attempting to add horizontal distance constraint as a driven constraint causes an immediate FreeCAD crash.  Problem does NOT occur with 0.20.2 release.  Problem also occurs at least as far back as 33411.

Steps to reproduce:
- Create Body
- Create Sketch
- Select XY-plane
- Select Centered Rectangle
- Drag to create any size rectangle
- Exit Rectangle tool
- Toggle Driving Reference (constraints in manu should now be blue)
- Select Horizontal Constraint
- Select one of the horizontal segments
- ** BOOM **

I actually found this by accident, as I _really_ was looking for the toggle for construction geometry, and grabbed the wrong icon.  Regardless, it shouldn't be a hard crash.

[image]

I could include a screen recording, but it is 140 MB.

### Anything else?

Here's the first part of the crash dump:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [35814]
Path:                  /Applications/Other Apps/FreeCAD-rc1.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [35808]
User ID:               501

Date/Time:             2023-08-01 11:56:23.4280 -0500
OS Version:            macOS 13.5 (22G74)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       947C7752-34E1-46BB-AE1A-6BF3FE48F4D6

Time Awake Since Boot: 360000 seconds
Time Since Wake:       95792 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x0000000100e6127c
Exception Codes:       0x0000000000000002, 0x0000000100e6127c

Termination Reason:    Namespace SIGNAL, Code 10 Bus error: 10
Terminating Process:   exc handler [35814]

VM Region Info: 0x100e6127c is in 0x100e18000-0x100ed4000;  bytes after start: 299644  bytes before end: 470403
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      mapped file                 100e14000-100e18000    [   16K] r--/rw- SM=COW  ...t_id=4a6bbccb
--->  __TEXT                      100e18000-100ed4000    [  752K] r-x/rwx SM=COW  ...c++.1.0.dylib
      __DATA_CONST                100ed4000-100edc000    [   32K] r--/rw- SM=COW  ...c++.1.0.dylib

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libc++.1.0.dylib              	       0x100e6127c std::__1::moneypunct<char, false>::do_curr_symbol() const + 0
1   libc++.1.dylib                	       0x186881a38 std::__1::basic_ostream<char, std::__1::char_traits<char>>::operator<<(unsigned long) + 404
2   SketcherGui.so                	       0x164443d38 void boost::io::detail::put<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&>(boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&, boost::io::detail::format_item<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>::string_type&, boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>::internal_streambuf_t&, std::__1::locale*) + 596
3   SketcherGui.so                	       0x164443a14 void boost::io::detail::distribute<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&>(boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char>>&, boost::io::detail::put_holder<char, std::__1::char_traits<char>> const&) + 180
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-01:
You are talking about "Constrain horizontally", not "Constrain horizontal distance" (dimension), right ? When you use "Toggle driving/reference constraint", only the dimension constraints become blue. Anyway, nothing wrong happens when I follow those steps and apply one of the aforementioned constraints to the geometry.

You could share a gif showing what happens.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33658 (Git)
Build type: Release
Branch: master
Hash: ba5f7e6d75dee09ec7076ad24895b04e56f21b63
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

JohnOCFII, 2023-08-01:
> You are talking about "Constrain horizontal distance" (dimension), not "Constrain horizontally", right ?

Correct.

JohnOCFII, 2023-08-01:
> You could share a gif showing what happens.

(Long rathole of how to create a GIF on my Mac...) :)

[image]

Syres916, 2023-08-06:
Cannot replicate on Linux.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33705 (Git)
Build type: Release
Branch: master
Hash: a4929211d43ff35bd4fea7833c0333ab17c2ba1b
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.3
  * A2plus
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * freecad.gears 1.0.0
  * Curves 0.6.13
  * toSketch 1.0.1
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.56
  * sheetmetal 0.2.63
```

yangit, 2023-08-19:
I confirm having this issue on Apple M1 Macbook pro under MacOs 13.5 (22G74) in Freecad version:
I think I have downloaded apple silicon version of freecad.
[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]

I can not create any driven constraints, it crashes the same way as on gif above.

bartgrantham, 2023-08-28:
```
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * sheetmetal 0.2.63
```

I am having a similar issue, and I had it on 0.20 as well.  Clicking on external geometry when creating constraints crashes FreeCAD.  Not always, though.  As you can hopefully see from my attached mp4 I was able to create circles that reference the external geometry of the outer edge of the part but when I go to add a fifth circle to the sketch it crashes.  Sorry I can't give a better description, I'm a FreeCAD newbie and unfortunately this bug has prevented me from moving past that level.

I've included a quick scan of the stack trace from Apple's debugging window.  If I were to make a wild guess it looks like a recursive function in an infinite loop running out of stack space.

https://github.com/FreeCAD/FreeCAD/assets/913646/df238aea-d432-4fab-9857-b8adecb90e4c

scandey, 2023-09-05:
I just want to +1 this issue for Mac, I can't consistently replicate but I do have one full crash report attached here. Not exactly the same as the one shown in the screencap above, but presumably related since it was the same cause.
[freecad0_21_crash_constraint.txt](https://github.com/FreeCAD/FreeCAD/files/12528677/freecad0_21_crash_constraint.txt)

JohnOCFII, 2024-02-13:
With the changes to constraints in the more recent 0.22 dev builds, I'm not able to dupllcate the issue.  Not sure if the underlying issue has been resolved, or just that the particular selection flow is now different such that the problem isn't likely to occur.  Would love some additional input/testing.

maxwxyz, 2024-06-24:
I cannot reproduce, so I'll close it.

closed 2024-06-24 by maxwxyz