FreeCAD/FreeCAD #10041: [Problem]

gilmiz, 2023-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33669 (Git)
Build type: Release
Branch: (HEAD detached at 0.21rc1a)
Hash: e5a00e153d36b0d53ebd568f4f47e95fbb5d0fcd
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Danish/Denmark (da_DK)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.56
  * freecad.gears-master 1.0.0
  * Materials
  * MeshRemodel 1.8919.0
  * Render 2023.2.6
  * sheetmetal 0.2.63
  * toSketch 1.0.1
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

Some lines, looks like arcs, are missing when inserting a projection group. See the attached image.
The same operation results without errors on FC 0.20.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-02:
Can you attach the FCStd file?

WandererFan, 2023-08-02:
Also a marked up picture showing where the missing bits should be, please.

gilmiz, 2023-08-03:
Updated image with missing lines and source file added.

WandererFan, 2023-08-03:
The Direction and XDirection for the views are off a bit which is causing those segments to be projected as ellipses with very small diameters.
`Front
>>> obj.Direction
Vector (2e-06, -2e-06, 1.0)
>>> obj.XDirection
Vector (4e-12, 1.000000000004, 2e-06)
>>> 

Left
>>> obj.Direction
Vector (-4e-12, -0.9999999999980002, -2e-06)
>>> obj.XDirection
Vector (2e-06, -2e-06, 0.9999999999960002)`
To fix it, update these properties to be exactly the correct values (like (0.0, 0.0, 1.0).

WandererFan, 2023-08-03:
[image]
[image]

luzpaz, 2023-08-05:
Maybe techdraw should somehow highlight the negative floating point number or offer a hazard emoji next to it with a hover tooltip explaining the possible warning ?

WandererFan, 2023-08-05:
A negative number is perfectly acceptable (ex (0, 0, -1) indicates view from the bottom)

"-0.00" is a clue that the real number is something like "-0.00000123".

The real fix might be for Gui::View3DInventorViewer::getViewDirection() to not return garbage after the sixth or seventh decimal place.

Don't know how we would tell the PropertyEditor vector updater that TD Direction vector should be handled differently.

closed 2023-08-05 by WandererFan

luzpaz, 2023-08-05:
> Don't know how we would tell the PropertyEditor vector updater that TD Direction vector should be handled differently.

@WandererFan open a ticket for this ?

WandererFan, 2023-08-08:
I'd rather get the right value from Coin for the Direction.