FreeCAD/FreeCAD #10042: [Problem] Imported BREP file displayed incorrectly on newly installed FreeCAD

Roy-043, 2023-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git)
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```
```

### Subproject(s) affected?

Core

### Problem description

The attached BREP file is displayed incorrectly when imported if FreeCAD has been started without a user.cfg file.

Steps:
1. Close FreeCAD, delete or rename user.cfg, and restart.
2. [Std_Open](https://wiki.freecad.org/Std_Open) the BREP file.
3. The shape appears as the projection of half of a cylindrical face without edges.
4. Open the [Preferences editor](https://wiki.freecad.org/Preferences_Editor).
5. Press "Apply".
6. Press "OK".
7. The shape is now shown correctly.

This is a cleaned up version of #10021.

ZIP file containing the BREP file:
[facePiecesCannotTriangle9_9_0.zip](https://github.com/FreeCAD/FreeCAD/files/12238892/facePiecesCannotTriangle9_9_0.zip)

Wrong display:
![brep_wrong_display](https://github.com/FreeCAD/FreeCAD/assets/70520633/8256a2af-5e0d-4859-95b1-44c697974c79)

Correct display:
![brep_correct_display](https://github.com/FreeCAD/FreeCAD/assets/70520633/7edd92f8-3cd8-4413-b20e-d5d6745259c7)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

StromanStroman, 2023-08-03:
What's the difference between the correct face and the wrong face in the previous version freecad that has not been reset?Is there any problem in the face itself?What's the problem?

Roy-043, 2025-01-17:
This is  still an issue. Instead of renaming the CFG file you can now test in safe mode.

ElementW, 2026-09-13:
Got fixed somewhere between the 2026.01.14 and 2026.03.04 weeklies (AppImage; both using OCC 7.8.1 so it probably was an FC issue). Couldn't reproduce on anything after that.

@maxwxyz Can be closed I think.

closed 2026-09-14 by maxwxyz