FreeCAD/FreeCAD #10046: [Problem] Multimaterial and override align give wrong result

kaiwas, 2023-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * BIM 2021.12.0
  * CADExchanger
  * Curves 0.6.12
  * DynamicData 2.46.0
  * frame 0.1.0
  * FreeCAD_Info
  * IconThemes
  * Manipulator 1.5.0
  * parts_library
  * PieMenu 1.2.7 (Disabled)
  * PitchedRoof
  * QuickMeasure 2022.10.28
  * Reinforcement
[/code]
```

### Subproject(s) affected?

Arch

### Problem description

The combination of multi-material and override alignment in the wall settings gives strange results.

[image]
[image]
[image]
[image]
[image]
[Wall_MultiMat_OverAlign.zip](https://github.com/FreeCAD/FreeCAD/files/12244363/Wall_MultiMat_OverAlign.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-08-03:
Indeed it's problem :) @paullee0 I think this was your code... Mind to have a look?

paullee0, 2023-08-03:
Hmmm, if my memory is correct, 'overide-align' does not support Multi-Material' (at the moment)  :D

kaiwas, 2023-08-03:
> 
It seems I discovered an undocumented possibility. :D
Eh... if we could improve this feature and make another override offset, it would be just fine.

paullee0, 2023-08-04:
In fact, I remember I thought about it and found the 'Align' property would be complicated when there is Multi-Material.

User can simply edit the position of an edge to make the Wall location as user desire instead of using 'Align'; on the other hand the 'direction' of layer is determined by the Sketch edge's 'direction'.  So it would rather be more simpler to edit the Sketch edge to achieve the effect.

Or there are other thoughts?

For 'Override Offset', maybe you open another post at the Forum to discuss the idea and see if there are other users would join the discussion :D

kaiwas, 2023-08-04:
> For 'Override Offset', maybe you open another post at the Forum to discuss the idea and see if there are other users would join the discussion :D

I will create a large topic for discussing the construction of scenes.
I will point out the current difficulties and possible changes that, in my opinion, would make the walls better and more convenient.
But in addition to the walls, sketches (lines) can still be affected, because. walls are built on top of them.
I'll try to make my thoughts as clear as possible.

paullee0, 2023-08-04:
Yes please :D

luzpaz, 2023-10-24:
bumping for status

luzpaz, 2023-10-27:
> I will create a large topic for discussing the construction of scenes.

@kaiwas where is this topic ?

kaiwas, 2024-03-02:
> @kaiwas where is this topic ?

I haven't done it yet. There are too many features that I want to mention. I am collecting information. It turned out to be more difficult than I expected)

luzpaz, 2024-03-18:
@paullee0 can you link to forum thread that discusses this ?

kaiwas, 2024-03-18:
> link to forum thread that discusses this ?

You can try discussing multimaterials in this thread
https://forum.freecad.org/viewtopic.php?t=86010

In addition, I learned the pleasant ability to calculate the volume of each layer. This means that working with multimaterials becomes more convenient and relevant. More details in this thread.

https://github.com/FreeCAD/FreeCAD/issues/12900

kaiwas, 2025-01-13:
> Hi! This issue hasn’t seen activity in a while. We automatically check each issue after 190 days without activity to keep the backlog tidy. If it’s still relevant, please update to the latest [...]

The problem is still relevant

kaiwas, 2025-09-08:
> For avoidance of doubt what is user expected result, could make a simple sketch?

There is no solution to this problem. And it is not about erroneous indents. All this is because the program cannot know which side is external and which is internal.

Ideally, a multi-layer wall could look like this. But given the current implementation, this is probably impossible.

[image]

paullee0, 2025-09-08:
Good diagrams for inclusion on wiki.

It seems doable.

kaiwas, 2025-09-08:
> for inclusion on wiki.

Wiki? It's impossible to arrange layers like that now.
For the wiki, I would make the colors more calm. Let me know if the wall layers start to behave like that and I'll put together a new picture for the wiki.

kaiwas, 2026-03-19:
still relevant