FreeCAD/FreeCAD #10048: [Problem] QWaylandShmBuffer: mmap failed (Invalid argument)

sdifru, 2023-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: makepkg
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * dodo 1.0.0
  * FreeCAD_assembly3 0.12.0
  * FreeCAD_FastenersWB 0.4.58
  * FreeCAD_Assembly4 0.50.2
  * FreeCAD_SheetMetal 0.2.63
  * Lattice2 1.0.0
  * freecad.gears 1.0.0
  * CurvesWB 0.6.12
[/code]
```

### Subproject(s) affected?

None

### Problem description

I installed the freecad from arch linux repository. I use [sway](https://github.com/swaywm/sway) as the window compositor.

When I opened the 0.21 freecad, it shows this on the report view:
`(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()`

Then I opened a file. I rotated the object view using mouse and report view shows this:
`QWaylandShmBuffer: mmap failed (Invalid argument)`

It interrupts the workflow because the report view pop up every time I rotate the object view.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-04:
Found a similar looking bug over at https://bugs.kde.org/show_bug.cgi?id=429967#c3

luzpaz, 2023-08-21:
@sdifru can you try the Qt5 solution solution in https://github.com/githubuser0xFFFF/Qt-Advanced-Docking-System/issues/288#issuecomment-1213371312

> > A work-around  (in Qt5) is to set `XDG_SESSION_TYPE`. For example:
> > ```shell
> >  XDG_SESSION_TYPE=x11 /path/to/FreeCAD
> > ```
> 
> This fix doesn't work when compiling with Qt6. Instead, you should use `QT_QPA_PLATFORM=xcb`

sdifru, 2023-08-23:
@luzpaz The workarounds work. Should I close this issue or let this opened until it is fixed?

luzpaz, 2023-08-23:
xyproto, thanks for being so 'johnny on the spot'! 

Hi @adrianinsaval, should we address this on our end or defer to downstream ?

adrianinsaval, 2023-09-21:
setting XDG_SESSION_TYPE is not a true fix because it just makes freecad run under xwayland. I cannot reproduce the mmap error anymore on 0.22 and we have another issue for the requestActivate error: #10369 so I'm leaning towards closing this one. @sdifru could you confirm that the mmap error is no longer present?

sdifru, 2023-09-25:
@adrianinsaval I don't know if we can say that this workaround solves the issue but it makes the issue doesn't appear anymore.

3x380V, 2024-01-19:
@adrianinsaval I'm still getting `QWaylandShmBuffer: mmap failed (Invalid argument)` with today's git version (Debian 12, built against Qt6).

PR #14547 (2024-07-29): Gui: Buggy Wayland custom cursors workaround
Fixes #10048

Until Qt 6.6 custom cursors are not shown at all on Wayland, so add a workaround. See also QTBUG-95434.
Workaround is limited to FreeCAD running on Wayland and Qt versions between 5.13 and 6.6. Versions prior implementing needed `QImage:: convertTo` are unlikely to be found running on Wayland.
As soon as the oldest supported Qt version becomes 6.6 on Linux, this workaround should be removed.
closed 2024-07-29 by chennes