FreeCAD/FreeCAD #10050: What's the reset have done?

StromanStroman, 2023-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

I just want to know the process and reason behind steps you show in [#10042](https://github.com/FreeCAD/FreeCAD/issues/10042).
Why reset ,apply preference,can make incorrect display correct?
Is problem of freecad,or is problem of the face itself,or is problem of occ apis?
I am now quite puzzled,please make me clear,thank you.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

StromanStroman, 2023-08-03:
What reason cause those problems?

Roy-043, 2023-08-03:
You should not have created another issue to address your problem. Please do not do this again. Thanks.

Closing this as duplicate.

closed 2023-08-03 by Roy-043