FreeCAD/FreeCAD #10051: [Problem] Could not create pixmap from qss:images_dark-light/background_freecad_ExtremProDark.svg

sdifru, 2023-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: makepkg
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * dodo 1.0.0
  * FreeCAD_assembly3 0.12.0
  * FreeCAD_FastenersWB 0.4.58
  * FreeCAD_Assembly4 0.50.2
  * FreeCAD_SheetMetal 0.2.63
  * Lattice2 1.0.0
  * freecad.gears 1.0.0
  * CurvesWB 0.6.12
[/code]
```

### Subproject(s) affected?

None

### Problem description

`Could not create pixmap from qss:images_dark-light/background_freecad_ExtremProDark.svg`

The message shows up on report view right after I open freecad. I don't know what to do with it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-08-03:
I bet that's a typo (assuming you copied the message?) -- it should probably be referencing "qss:images_dark-light/background_freecad_Extrem**e**ProDark.svg" (with an "e" at the end of Extreme). @MisterMakerNL ?

sdifru, 2023-08-03:
@chennes You are right. I copied pasted the message.

[image]

luzpaz, 2023-08-17:
This is a quick fix. Lets just get it done?

MisterMakerNL, 2023-08-18:
This PR should be first https://github.com/FreeCAD/FreeCAD/pull/10238#issuecomment-1682572139
After that I'll make a new cleaner pr.

MisterMakerNL, 2023-09-05:
This been fix with the theme pr. Don't have an exact number at hand tho.

closed 2023-09-07 by luzpaz