FreeCAD/FreeCAD #10055: [Feature]

magnificu, 2023-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33682 (Git)
Build type: Unknown
Branch: master
Hash: 9dce2525f8efe0873787f4388f10e11e168fe16b
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly3 0.12.0
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

None

### Problem description

I would like to propose an easy improvement in the Sketcher module that can make it easier to use and faster.

While using the following button commands: "Constrain horizontal distance" & "Constrain vertical distance"

It would be great to run highlight edges from "Create external geometry" when the mouse hovers above external geometries and in case the use click to create the external geometry automatically.

I think all the functionality is existing already.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-03:
Please add some title to the [Feature] label. Something like "Automatic projection of external geometry for dimensions", for instance.

NXTler, 2023-08-17:
Yes, we need this one

Ely31, 2023-10-15:
Yeah Fusion 360 does this and it is nice. It does get annoying though when sometimes there's a lot of stuff behind the sketch that you can't even see and you can end up accidentally grabbing that geometry. Fusion solves this by disabling snapping, auto constraints and auto geometry projection when you hold down the control key. That's out of the scope of this feature, but I wanted to mention it. 90% of the time auto projection is awesome though, maybe just add a check box for it in the sketcher preferences in case some people find it annoying.

luzpaz, 2023-10-15:
Feel free to rally support an interest for this feature on the forum or discord. There are many fronts we are tackling ATM but this feature seems like a nice UI/UX feature. If someone is willing to code it... ?

Roy-043, 2026-11-14:
Implemented with #31916.

closed 2026-11-14 by Roy-043