FreeCAD/FreeCAD #10057: [Problem] App Crashes while Transforming an Imported Image

tootzreyes, 2023-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
FreeCAD_0.21.0-2023-08-02-conda-macOS-x86_64-py310.dmg. This version is given by report/his reply to my facebook report that Mac version 0.21 does not work.

Chris Hennes
Author
Thanks - can you test the bundle from here? https://github.com/Fre.../FreeCAD-Bundle/releases/tag/0.21.0
```

### Subproject(s) affected?

None

### Problem description

I have a project which has an imported image, for reference. With the version 0.21, the transform function does not work and immediately crashes the app on arrow dragging / movement. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-08-03:
Looks like another duplicate of #9465.

luzpaz, 2023-08-13:
@tootzreyes I'm closing this in favor of #9465

closed 2023-08-13 by luzpaz