FreeCAD/FreeCAD #10058: Spreadsheet prepends single quote to cell string value

MattN92705, 2023-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.21.0.33682 (Git)
Build type: Release
Branch: master
Hash: 9dce2525f8efe0873787f4388f10e11e168fe16b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * ExtremeProLight 1.6.0
  * Help 1.0.3
  * Assembly4 0.50.2
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Problem description

FreeCAD weekly build 33682 has a bug in Spreadsheet when entering text into a cell.

- Type some text into a cell, e.g., “This is text”
- Hit return to enter the value
- The value appears to be correct when looking at the text in the spreadsheet, but…
- Select the cell with the text and look at the “Content” line at the top
- The cell’s content is actually “ ‘This is text “ (note the extra single quote prepended to the string).

Entering numbers seems to work correctly.

I like to build lists of “parameter name” : “value” pairs in adjacent cells by copying the parameter name and pasting it into the Alias property of the cell with the value.  The single quote makes the parameter name an invalid alias, requiring manually deleting the single quote from the alias field.

Small bug, but if you are putting together a large spreadsheet with parameters, it gets tiring.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2023-08-03:
Interesting.  I saw this change, and since it impacted my use of the EasyAlias macro, I created a note in the forum for the macro author, since I wasn't sure if this changed behavior was actually intended with 0.21 or not.

https://forum.freecad.org/viewtopic.php?t=80122

chennes, 2023-08-03:
It is intended behavior, to take care of some string-handling issues. Isn't there a macro that makes these aliases for you? I feel like I have seen that in the addon manager.

MattN92705, 2023-08-03:
this is a recent change?  it certainly didn't used to behave that way.

While i understand that this may be there to make string handling easier, it seems like a very strange behavior from a user perspective.

I honestly haven't looked at alias macros - but if the spreadsheet is going to stay this way, i will.

chennes, 2023-08-03:
The fix is probably to adjust the paste code -- that's how Excel handles it, basically. When you paste from within Excel the explicit string indicator is retained when pasting into a cell, but dropped when pasting anywhere else.

JohnOCFII, 2023-08-03:
The EasyAlias macro seems quite popular.  I'm hoping the author (TheMarkster, in the forums) sees my note and can make the update.

JohnOCFII, 2023-08-04:
FYI - TheMarkster has posted the updated code for his macro (tested, works well) in the Forum thread, and will soon had the updated EasyAlias Macro posted so the Addon Manager will find it.

https://forum.freecad.org/viewtopic.php?t=80122

luzpaz, 2023-08-05:
Close ticket ?

MattN92705, 2023-08-05:
IMHO a macro with a workaround for a specific use case is not a fix. As long as the contents of a cell are being modified by the software and cannot be cut/pasted as expected, i would still consider it a problem that needs correction. Go ahead an name some other software that behaves that way…

hermannjt, 2023-08-10:
This problem actually extends beyond the Spreadsheet workbench. My primary use of Freecad is for cabinet making and extensively use the A2P Parts Infor and Partslist features to create cut lists for my different cabinet assemblies. Once I have the parts Lists, I copy those to an Excel APP I developed for cut lists to send to my supplier(s) for quotes. 

Needless to say, having '&apos; appended to the front of text strings (see image below) is a major issue for users like myself!

[image]

Jodaille, 2023-08-12:
Hello, I agree with @hermannjt it is a strange behaviour, 
we could not copy/paste content of spreadsheet cell anymore,
in the alias field nor in another place (terminal, browser).

[image]

[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git) Snap 727
Build type: Release
Branch: tag: 0.21.0
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]

regularfry, 2023-08-14:
I'm just going to leave another "this is surprising behaviour that feels like a bug" comment - specifically, copy a field; paste it into the alias box; hit enter: that workflow is now broken.

MattN92705, 2023-09-14:
There hasn't been any "design" comments since @chennes said it was an intentional design decision.  Every time i use the spreadsheet I am dismayed to see this bizarre behavior - how can altering user-entered data be a good design?  As others have said, it is no longer possible to easily cut/paste between the spreadsheet and *any* other application.  I'd like to see some sort of acknowledgement from the development team that this needs to be fixed...

PAS-DRAFTER, 2023-10-02:
This behavior wasn't there before upgrading to v0.21.  It is quite annoying.  In the past, when I would type an equation into the cell and it was an error, it would turn red and make sure you fixed it.

Now it just puts a quote in front of it and accepts it.  Then when you figured out you have an error, now you have to go in and delete the quote before you can accept the change.

The ONLY advantage I see so far is that cell errors are quoted out so your spreadsheet no longer has errors when an equation changes.  This allows the calculations in the 3D side to continue.  However, if any of those drawing parts are using the spreadsheet with the errors masked now, it still causes grief, so I don't think the benefits outweigh the negatives here.

Another behavior that I noticed has changed is that if you change the Alias name in the spreadsheet now and you were using that Alias as a constraint or other variable, when you go and look at the link in that constraint, the <<Actual Name of Spreadsheet>> part of the link is not just <<Spreadsheetxxxx>> (whatever internal name FreeCAD uses).  I have to go in and reset the names to get the links to work again.

Not sure if this is related or a completely different issue.

ModuloFS, 2023-11-30:
> There hasn't been any "design" comments since @chennes said it was an intentional design decision. Every time i use the spreadsheet I am dismayed to see this bizarre behavior - how can altering [...]

Had the same problem today by using the Alias Manager Macro which does not work anymore because of that, costs me at least an hour to figure out what is happening! Not a very satisfying user experience...

jwueller, 2024-05-13:
It seems that this just isn't implemented fully. It's correct to be able to escape values with the leading `'`. It's like a formula starting with `=`. But the pasted value always including it is wrong. Copying it has two distinct use-cases:

1. Pasting the _value_
2. Pasting the _formula_

Basically, the copy operation needs to store both, in different [clipboard formats](https://learn.microsoft.com/en-us/windows/win32/dataxchg/clipboard-formats). This is supported on all major platforms as far as I'm aware.

Then, when pasting, the target application decides which format to use. Most will probably default to the plain text clipboard, which should contain the plain value only, excluding the leading `'`. The formula clipboard format would only be accepted in contexts that actually understand it, like spreadsheet cells.

Since FreeCAD formula rules aren't compatible with other software, even if they look similar, it probably should be a [private clipboard format](https://learn.microsoft.com/en-us/windows/win32/dataxchg/clipboard-formats#private-clipboard-formats).

See [QClipboard](https://doc.qt.io/qt-6/qclipboard.html).

Glitchy-Tozier, 2025-01-15:
Still relevant.

MattN92705, 2025-01-15:
Still present in 1.1.0dev revision 39841.

This hasn't seen much love from those who are able to do something about it, but IMHO it's a pretty strange behavior, especially for anybody who uses spreadsheets (none of Excel, LibreOffice/OpenOffice, Apple Numbers do this).  And, also IMHO, little "something weird" things like this are the enemy of bringing in new users.

luzpaz, 2025-01-25:
This pollutes input and impacts Spreadsheet workflow. Elevating to higher priority, Looks like @jwueller gave a diagnosis in https://github.com/FreeCAD/FreeCAD/issues/10058#issuecomment-2107964928

closed 2025-02-27 by chennes

MattN92705, 2025-03-02:
This fix works for cutting/pasting, so very good.  however, it doesn't address the issue of the cell contents showing in the prepended apostrophe in the "Content:" field.   And when editing the cell contents using the Content: field, you still have to deal with it.

That still strikes me as unexpected behavior for anybody used to using spreadsheets.

[image]

PR #19781 (2025-02-27): [Spreadsheet] Remove prepended apostrophe if cell content is copied into clipboard
This commit removes an apostrophe if it is the first symbol in the cell content when it is copied to clipboard.

Relevant PR is #9667. Perhaps, @0penBrain might want to comment this PR.

Closes #10058.
commit 4d4073c0 (2025-02-27): [Spreadsheet] Remove prepended apostrophe if cell content is copied into clipboard (#19781)

* [Spreadsheet] Remove prepended apostrophe if cell content is copied into
clipboard

Closes #10058.
commit 9bd72d7b (2025-02-27): [Spreadsheet] Remove prepended apostrophe if cell content is copied into clipboard (#19781)

* [Spreadsheet] Remove prepended apostrophe if cell content is copied into
clipboard

Closes #10058.