FreeCAD/FreeCAD #10061: [Problem] FreeCAD Crashes when moving a body into another.

TyPosey, 2023-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
  * Assembly3.backup1681771100.992021 0.11.4
  * CADExchanger
  * Manipulator 1.4.9
  * IconThemes 0.12.5
  * Design456 0.0.1
  * ExtremeProDark 2.6.4
  * ExtremeProLight 1.4.3
  * A2plus.backup1681771090.550049 0.4.60h
  * Assembly4 0.12.5
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Create two bodies.  Then move them to into one another to perform boolean.  FreeCAD crashes.

https://github.com/FreeCAD/FreeCAD/assets/25179104/e42170a1-fcd3-49da-b83e-12ba725e99e8

### Anything else?

Thanks for all you guys do in developing the software.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-08-03:
Duplicate of #9465.

chennes, 2023-08-04:
Most likely. @TyPosey can you successfully move just a single body, or does the move crash when you click OK?

TyPosey, 2023-08-07:
crashes when you click ok

chennes, 2023-08-09:
OK, this should be fixed in the Release version of FreeCAD (note that it is *only* fixed in Release, not in any of the ongoing developer builds -- it's a problem with the Coin library on Conda).

closed 2023-08-09 by chennes