FreeCAD/FreeCAD #10063: [Problem]

voxel3, 2023-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * fasteners 0.4.52
  * Help 1.0.3
  * IconThemes
  * MacroStartup
  * ShortCuts
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Part_SectionCut does not work if the view is in perspective.
It does not make any cuts and generates the following error:
`SectionCut error: the CompoundBoundingBox is empty`
It happens in version 0.20 as well as in 0.21.

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/55639245/0289210e-9528-45f2-8b8c-2f77165ce0cb

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-04:
I can confirm this.

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]

FOSCODER, 2023-08-04:
confirmed on windows 11
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Installed mods: 
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * Behave-Dark-Colors 0.1.1
  * BIM 2021.12.0
  * btl
  * CfdOF 1.24.3
  * Curves 0.6.13
  * Defeaturing 1.2.0
  * dodo 1.0.0
  * Dracula 0.0.4
  * Estimate 0.1.2
  * ExplodedAssembly
  * ExtremeProDark 2.9.0
  * ExtremeProLight 1.6.0
  * fasteners 0.4.59
  * freecad.gears 1.0.0
  * GDML 2.0.0
  * Help 1.0.3
  * InventorLoader 1.4.0
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
[/code]

FEA-eng, 2023-08-07:
@donovaly Can you take a look at this ?

closed 2023-08-09 by wwmayer

voxel3, 2023-08-10:
Great! Thank you so much.

commit a8a730bf (2023-08-09): fixes #10063: Part_SectionCut does not work if the view is in perspective
commit 419d5338 (2023-11-09): fixes #10063: Part_SectionCut does not work if the view is in perspective