FreeCAD/FreeCAD #10071: Group meshing for analysis is set to true in FEM General Preferences. Are you really sure about this? You could run into trouble!

Alberto4343, 2023-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

When trying to create a FEM mesh via the "FEM MeshGmshFromShape" tool this message appears:

"Group meshing for analysis is set to true in FEM General Preferences. Are you really sure about this? You could run into trouble!"

When i go to FEM General Preferences that option is grayed out and activated.

I'm having troubles to create a FEM mesh via the "FEM MeshGmshFromShape" tool and maybe this is the reason.

Excellent program and initiative. Thank you so much.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FOSCAS, 2023-08-04:
On the wiki it is shown greyed out and unchecked. After checking I have the same problem

Alberto4343, 2023-08-04:
I hope they check this issue soon, ¿Do you know how much do we have to wait?

luzpaz, 2023-08-05:
To raise probability of this issue getting addressed please open a thread in the FEM subforum https://forum.freecad.org/viewforum.php?f=18 (x-post link back to ticket)

Alberto4343, 2023-08-06:
@luzpaz, i can't register, when chosing my language it sends me to the previous page. ¿Do you know where to report that problem?

luzpaz, 2023-08-06:
What are you seeing ?

Alberto4343, 2023-08-06:
From this page https://forum.freecad.org/ucp.php?mode=register it sends me to this page https://forum.freecad.org/ucp.php?mode=register

Roy-043, 2023-08-06:
You can change this preference manually:

Start the [Std_DlgParameter](https://wiki.freecad.org/Std_DlgParameter) command:
Tools > Edit parameters

Go to the correct group:
BaseApp > Preferences > Mod > Fem > General

Change `AnalysisGroupMeshing` to `false`

Alberto4343, 2023-08-06:
It worked thank you. have you repaired it in the program?

Roy-043, 2023-08-08:
Issues are closed when they get fixed. I am not involved with the FEM code, so somebody else would need to step in.

Alberto4343, 2023-08-08:
I don't know well how this work, when we post an issue does it get to the developer team somehow or not?

luzpaz, 2023-08-16:
@Alberto4343 there are many many tickets open (897 at the moment) and we are all volunteers here. It's not instant fix just because it was reported.

commit 4932f17d (2024-02-24): Fem: Enable checkbox for references mesh groups - fixes #10071 (#12437)
closed 2024-02-24 by chennes