FreeCAD/FreeCAD #10073: [Problem] 0.21 Stable switching camera view modes no longer works with Spaceball buttons

glabifrons, 2023-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.2 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.59
  * 3D_Printing_Tools
  * A2plus 0.4.60n
  * Curves 0.6.13
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

Spaceball buttons can no longer switch the camera view modes in 0.21 (works in 0.20 and 0.19).
Other actions I've tested do seem to work properly (after reassignment, as all were lost during the upgrade).
(Note: I selected 0.21 Development as there is no option for 0.21 Stable in the Issue form)

I dug into the sources (disclaimer: I'm not a dev) and it appears that two functions (ViewPerspectiveCmd & ViewOrthographicCmd) were deleted from src/Gui/NaviCube.cpp from 0.20 to 0.21.
https://github.com/FreeCAD/FreeCAD/blob/fcad80494f8937a98be6b2d8fc45c54c930b7817/src/Gui/NaviCube.cpp#L1714
https://github.com/FreeCAD/FreeCAD/blob/fcad80494f8937a98be6b2d8fc45c54c930b7817/src/Gui/NaviCube.cpp#L1734
I couldn't find these in the 0.21 sources.

I'm only opening one report as I'm guessing the other problem is a side-effect of this removal: Spaceball button configs are cleared when going from 0.20 to 0.21 (repeatable). If this is not the case, I can open another issue (if needed).

Details:
I noticed that the Spaceball button configuration didn't carry over from 0.20 (I just upgraded to 0.21 today using the AppImage), so I proceeded to redefine the buttons. I found that while the buttons get programmed and issue commands, the two commands issued to change view modes to Perspective or Orthographic do not work.
I experimented in the Python Console and found that when I change the passed numeric parameter from a 0 to a 1 (and removed the hash), it changes modes properly.
Example of output when tapping the two buttons:
```
>>> # Gui.runCommand('Std_OrthographicCamera',0)
>>> # Gui.runCommand('Std_PerspectiveCamera',0)
```
If I manually enter either of the following, it does properly switch the view mode:
```
>>> Gui.runCommand('Std_PerspectiveCamera',1)
>>> Gui.runCommand('Std_OrthographicCamera',1)
```
I don't pretend to know what the numeric parameter is (help(Gui.runCommand) just shows it's the "index of the child command", not sure what that means), it's just the first thing I tried on a hunch.

After recovering my previous user.cfg and restarting 0.20, I captured a different set of commands in the Python Console (the view switching was working properly here):
```
>>> ### Begin command ViewPerspectiveCmd
>>> Gui.activeDocument().activeView().setCameraType("Perspective")
>>> ### End command ViewPerspectiveCmd
>>> Gui.runCommand('Std_OrthographicCamera',0)
>>> Gui.runCommand('Std_PerspectiveCamera',1)
>>> ### Begin command ViewOrthographicCmd
>>> Gui.activeDocument().activeView().setCameraType("Orthographic")
>>> ### End command ViewOrthographicCmd
>>> Gui.runCommand('Std_OrthographicCamera',1)
>>> Gui.runCommand('Std_PerspectiveCamera',0)
```

Upon switching back to 0.21, my config was again wiped (this is repeatable).
It appears the commands ViewOrthographicCmd & ViewPerspectiveCmd no longer exist, so are replaced with something that only does half of the intended actions when the Spaceball buttons are redefined in via the GUI.
```
>>> ViewOrthographicCmd
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'ViewOrthographicCmd' is not defined
>>> ViewPerspectiveCmd
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'ViewPerspectiveCmd' is not defined
```
Note that the long commands actually still work when entered into the console:
```
>>> Gui.activeDocument().activeView().setCameraType("Perspective")
>>> Gui.activeDocument().activeView().setCameraType("Orthographic")
```

I tried manually changing user.cfg as follows, but it did not work.
```
          <FCParamGroup Name="9">
            <FCText Name="Command">Gui.activeDocument().activeView().setCameraType("Perspective")</FCText>
            <FCText Name="Description"/>
          </FCParamGroup>
...
          <FCParamGroup Name="11">
            <FCText Name="Command">Gui.activeDocument().activeView().setCameraType("Orthographic")</FCText>
            <FCText Name="Description"/>
          </FCParamGroup>
```

Thank you very much for your time and attention, and for such an amazing piece of software. :)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

numo68, 2023-12-21:
This seems to have happened in https://github.com/FreeCAD/FreeCAD/commit/8abd6d0870a2d87c88228d1a8026ce5a1e4d0921. @Jolbas: was there any particular reason for the removal, or should all the removed commands be added back, probably to src/Gui/CommandView.cpp ?

Jolbas, 2023-12-21:
The reason was the already existing Std_OrthographicCamera command that does the same thing and when I did the change I couldn't find it was used anywhere else than in the Navicube.cpp. I didn't understand it could be a part of a configuration.
Did you try change your user.cfg to:
```
          <FCParamGroup Name="9">
            <FCText Name="Command">Std_PerspectiveCamera</FCText>
            <FCText Name="Description"/>
          </FCParamGroup>
...
          <FCParamGroup Name="11">
            <FCText Name="Command">Std_OrthographicCamera</FCText>
            <FCText Name="Description"/>
          </FCParamGroup>
```

glabifrons, 2023-12-21:
@Jolbas Thanks for looking at this.
It only ended up as part of my configuration as I used the GUI configuration options (Tools->Customize->Spaceball Buttons) to select those menu items for those SpaceBall buttons.
I just tested your suggestion and now when I press the buttons, I get this in the Python console, and nothing happens (I pressed each button once):
```
>>> # Gui.runCommand('Std_PerspectiveCamera',0)
>>> # Gui.runCommand('Std_OrthographicCamera',0)
```

glabifrons, 2024-06-08:
Is there any possibility this will be reverted/fixed in the upcoming 1.0?
Failing that, is there a workaround I can hack into the user.cfg (the ones previously mentioned obviously don't work).
If it helps, I can test with the following products:
SpaceBall 5000 USB
SpaceBall 2003 (serial)
SpacePilot (rectangular model, not the Pro)

glabifrons, 2025-02-02:
Yes, bot, this is still broken, even in FreeCAD_weekly-builds-40041-conda-Linux-x86_64-py311.AppImage.

@Jolbas Since those two very small functions were actually used for something and removing them broke that functionality, can we please have the functions added back into the code?

glabifrons, 2025-05-31:
Yes, bot, this is still broken, even in FreeCAD_weekly-builds-41905-conda-Linux-x86_64-py311.AppImage