FreeCAD/FreeCAD #10075: 3MF files exported from FreeCAD don't work in PrusaSlicer

greggjaskiewicz, 2023-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60k
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

None

### Problem description

exciting that 3MF is now main export format. 
But when importing these files into PrusaSlicer (slic3r) - I get message like:
```The size of the object is zero
Object size from file KentEVs-Kent EV logo.3mf appears to be zero.
This object has been removed from the model```

I tried the same with multiple models. 

I don't think it's the PrusaSlicer, as it has been out for quite a while. 

Prusa slicer version 2.6.0

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

greggjaskiewicz, 2023-08-05:
I just noticed that some files work, others don't.

Attached is the file that does not work.

[Kent_EVs_fails_3mf.zip](https://github.com/FreeCAD/FreeCAD/files/12266710/Kent_EVs_fails_3mf.zip)

luzpaz, 2023-08-05:
@MisterMakerNL do you mind testing this issue ?

grubertechnic, 2023-08-05:
Does the silcer allow multiple meshes? Since you have 3 single solids in your model in a part container, they also exported as intersecting meshes. If you do a boolean union the 3 solids (Kent Borders Body, Text Body, base) and then export it, does then the silcer accept it?

greggjaskiewicz, 2023-08-05:
The whole point of formats like 3MF is to allow multiple solids, unlike STL. You can download prusa slicer and try it yourself, it's open source.

grubertechnic, 2023-08-06:
Please be explicit, IS multiple solids supported in the slicer? Because this 

> exciting that 3MF is now main export format

is not rue, FreeCAD have no main export format.

> can download

No, have no interest in it, just asking for triaging.

greggjaskiewicz, 2023-08-06:
Slicer does support it yes

MisterMakerNL, 2023-08-06:
The text and the path in the Freecad file gets exported as a surface. Since 3D builder also doesn't like it, I think surface object aren't supported for 3FM files.
[image]

The macro has no issues with it, but that exports it as STL file:
`import FreeCAD, FreeCADGui
import Mesh
import sys
import math
import os
import subprocess

SLICERAPP= "C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe" # Put your Slicer program here "C:\ProgramData\Microsoft\Windows\Start Menu\Programs\superslicer\superslicer.exe"
#SLICERAPP= "C:\Program Files\superslicer\superslicer.exe"

def getVisibleObjects(obj, objset):
    sobjs = obj.getSubObjects()
    if not sobjs:
        objset.add(obj)
        return [obj]
    res = []
    for sub in sobjs:
        sobj,parent,childName,_ = obj.resolve(sub)
        if not sobj or sobj in objset:
            continue
        objset.add(sobj)
        vis = parent.isElementVisible(childName)
        if vis < 0:
            vis = sobj.Visibility
        if vis:
            res += getVisibleObjects(sobj, objset)
    return res

# user need to set this directory where slicing software is located
OutDir = FreeCAD.ActiveDocument.FileName.replace(".FCStd", "--")
# Get Objects in document
doc = FreeCAD.ActiveDocument
stlFile = ""
stlFiles = [ SLICERAPP ]
# hide all
objs = FreeCADGui.Selection.getSelection()
if not objs:
    objs = []
    objSet = set()
    for obj in doc.RootObjects:
        if obj.Visibility:
            objs += [ o for o in getVisibleObjects(obj, objSet) if hasattr(o, 'Shape') ]

for obj in objs:
  stlFile = OutDir+str(obj.Label)+".stl"
  Mesh.export([obj],stlFile)
  stlFiles.append(stlFile)
  print ("Exporting " + stlFile + "\n")
print ("Calling subprocess: " + str(stlFiles)+"\n")
subprocess.Popen(stlFiles)
`

luzpaz, 2023-08-06:
@MisterMakerNL Thanks.  
BTW, I put the code in codeblocks.

wwmayer, 2023-08-09:
See also: https://github.com/FreeCAD/FreeCAD/issues/9731

wwmayer, 2023-08-09:
As another application you can use Blender and install the 3MF add-on from: https://github.com/Ghostkeeper/Blender3mfFormat
When doing so Blender is perfectly able to load the file. So, to me it sees it's a problem with PrusaSlicer (tested with v2.4.2).

MisterMakerNL, 2023-08-09:
> 

Then the issue is that it exports solids as surfaces!
And that indeed prusaslicer and 3D builder don't support surfaces.

greggjaskiewicz, 2023-08-09:
Latest prusa slicer is 2.6

wwmayer, 2023-08-09:
> Then the issue is that it exports solids as surfaces!
And that indeed prusaslicer and 3D builder don't support surfaces.

You're right. That's the problem that PS doesn't support surfaces.

In https://github.com/FreeCAD/FreeCAD/blob/3283107f38ea4cd3de402418548738d729196166/src/Mod/Mesh/App/Core/IO/Writer3MF.cpp#L145 FreeCAD explicitly checks whether the mesh is a solid (i.e. model) or a non-solid (i.e. surface).

What happens if PrusaSlicer loads a mesh that isn't a solid. Will this cause some printing issues?

grubertechnic, 2023-08-10:
> Slicer does support it yes

Thanks! being expict is here always helpful. For me it was ambiguous.

cooper66b, 2023-08-10:
I had the same issue today with FC 0.21.0, when I export the following part motorAdd 07b_01. to get complete information I used "add dependent objects to selection" and than
"export " with the following note "21:26:44  (qt.xkb.compose) failed to create compose table"
The import in PrusaSlicer 2.6.0 gave me 
"Objects size file motorAdd 07b_FC0.21.0.3mf appears to be zero These objects have been removed from the model"
When I do the same which a part which I modified only with 0.20.2 I don't have any problems.
So I switched back to FC 0.20.2. With the first import I get a long list with 
"No extension found of type 'PartGui::ViewProviderGridExtension'", which disappears after loading saved file. Export with this version seems to be ok.
If needed I can provide both exported files.

mw46d, 2023-08-25:
The problem is around the 3MF object type:-( The newer Writer3MF uses an object type 'surface' for things which it does not consider 'solid'. And the Prusa-Slicer does not want to handle objects of type 'surface' (and that blocks the import  completely:-( )

Unfortunately, extruded ShapeStrings (and probably other things) are not considered solids by the Writer3MF code:-( That's where all this falls apart. 

[image]

wwmayer, 2023-08-30:
> It seems to be only 'sometimes', the extruded ShapeStrings end up as 'surfaces'. I don't have a good idea yet, under which conditions:-(

The question is what happens if a mesh is declared as "model" even if it's not a solid? PS will be able to import the file but what happens when trying to print it?

mw46d, 2023-08-31:
I  think, PrusaSlicer (and most slicers) can handle that to a certain extend. At least my strings come out ok (modulo the attachment  issues with very small extrusions). 

[image]

The PVA+ is still not washed off , but otherwise pretty good;-) 
[image]

wwmayer, 2023-08-31:
In https://github.com/FreeCAD/FreeCAD/pull/10446 there is now an option to treat surface meshes as solid -- it's on by default.

khimaros, 2023-09-03:
is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/9731 ?

mw46d, 2023-09-03:
Very easy to check, unzip your *.3mf file and check the included 3D/3dmodel.model for objects of type 'surface'. The PrusaSlicer does not like those:-(

`marcow@feather6:/tmp/mw1$ unzip ~/3mf_022.3mf
Archive:  /home/marcow/3mf_022.3mf
  inflating: 3D/3dmodel.model        
  inflating: _rels/.rels             
  inflating: [Content_Types].xml     
  inflating: Metadata/thumbnail.png  
  inflating: Metadata/thumbnail1.png  
  inflating: Metadata/thumbnail2.png  
marcow@feather6:/tmp/mw1$ fgrep '<object' 3D/3dmodel.model
  <object id="1" type="model">
  <object id="2" type="surface">
  <object id="3" type="model">
marcow@feather6:/tmp/mw1$ 
`

closed 2023-09-06 by wwmayer