FreeCAD/FreeCAD #10076: [Problem] The trim tool fails to detect segments when circles and lines intersect

raindropsfromsky, 2023-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33658 (Git)
Build type: Release
Branch: master
Hash: ba5f7e6d75dee09ec7076ad24895b04e56f21b63
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/India (en_IN)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.60n
  * ArchTextures
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * btl
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * ExplodedAssembly
  * fasteners 0.4.59
  * freecad.gears 1.0.0
  * Glass (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * SearchBar 1.0.2
[/code]
```

### Subproject(s) affected?

None

### Problem description

The **Trim** tool fails to detect the intersecting parts in a sketch.
When I hover the trim Tool on a segment, it detects unexpected parts of the circles and lines, and deletes the wrong sections.

Full details of this problem are provided at the FC forum here:
https://forum.freecad.org/viewtopic.php?p=697519
A sample project file is attached in that post, and screenshots describe the problem in detail.

If we compare the experiences of different users, it is clear that the behavior of Trim tool is path-dependent.
(Different people ended up with different results based on the order in which they deleted the segments.)

### Anything else?

A veteran user @Domad has confirmed that the problem _does_ exist.
https://forum.freecad.org/viewtopic.php?p=697488#p697488
He used a workaround of breaking some segments, but that is a desperate measure (not an elegant solution!)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

raindropsfromsky, 2023-08-05:
The simplest test case is provided below:

In the structure below, I delete a segment from the middle circle with Trim tool.
After that, I hover the Trim tool over other segments of the same circle.
But the Trim tool fails to detect the segment under the mouse pointer.
It continues to show some other segment.

This is 100% repeatable.
[image]

raindropsfromsky, 2023-08-05:
I have created the simplest test case that is 100% repeatable.

[image]

Once the Trim tool deletes a single segment from the circle, it fails to detect the two segments that lie between the three vertical lines. When the mouse is hovered over the other segments, the tool is able to detect them.

Syres916, 2023-08-05:
Another workaround is to temporarily disable _Snap to Objects_ https://wiki.freecad.org/Sketcher_Snap while using the Trim tool. A more elegant solution should still be investigated.

raindropsfromsky, 2023-08-05:
Yes, I can confirm that if the **Snap to object** option is turned off, the **Trim** tool can detect all segments.

But this factor should not influence the Trim tool's behavior at all!

raindropsfromsky, 2023-08-05:
Linking **Snap to object** option with the **Trim** tool is neither logical nor intuitive!

- The option should snap-constrain new points to the element over which the mouse is hovering. This option is apparently unrelated to the Trim tool.
- Even when the snap option is active, the Trim tool detects smaller segments but misses the larger ones!
- The tooltip for the **Snap to objec**t option says that _it snaps to an object that is preselected_. However, in our case, (a) nothing is preselected, and (b) we are not trying to snap a point to an object.
- An object is preselected when we click on it beforehand. When we drag a point on any object, it _does_ get highlighted, but we cannot call it "preselected".

maxwxyz, 2024-03-03:
Is this still present in the weekly builds?

maxwxyz, 2024-03-09:
@raindropsfromsky I cannot reproduce on the current weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Please let me know if this bug exists for you, otherwise I'll close this issue.

closed 2024-03-30 by maxwxyz