FreeCAD/FreeCAD #10077: [Problem] Pulldown menus have a transparent background in dark and light themes

Rexbas, 2023-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 41b058e)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Problem description

In 0.21.0 the background of pulldown menus are transparent making it hard to read. This happens with the dark and light theme, not with classic theme.
[image]
This happens when running Linux but for Windows 11 this problem does not occur.

### Anything else?

Operating System: KDE neon 5.27
KDE Plasma Version: 5.27.7
KDE Frameworks Version: 5.108.0
Qt Version: 5.15.10
Kernel Version: 6.2.0-26-generic (64-bit)
Graphics Platform: X11
Processors: 16 × 12th Gen Intel® Core™ i5-12600K
Memory: 15,4 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2
Manufacturer: Gigabyte Technology Co., Ltd.
Product Name: Z690 UD DDR4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Rexbas, 2023-08-05:
I just build 0.21.0 on Linux and now it works fine. I guess this problem only occurs in the flatpak. The beta flatpak worked fine until and including version 0.21.0.33439.

```
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33439 (Git)
Build type: Release
Branch: (HEAD detached at 976ecb3)
Hash: 976ecb3754700dfe6059699b77f896f9610891e3
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.7.1
Locale: English/United States (en_US)
```

luzpaz, 2023-08-05:
@adrianinsaval please transfer to flatpak repo, TIA

Rexbas, 2023-08-05:
When I change the background-color on this line to a color instead of transparent it works fine. Not sure if this also affects something else. @MisterMakerNL should we change this to a color or does this break the style elsewhere?
https://github.com/FreeCAD/FreeCAD/blob/aa535962896a411e06f72a235ca58db0002907db/src/Gui/Stylesheets/Darker-blue.qss#L121

MisterMakerNL, 2023-08-05:
There was a similar issue with the python console pop-up and I fixed it in the addon. I hope this is the same issue, although really weird it somehow uses the wrong class in the flatpack....
I will try to push everything to a PR.
Please try the beta version in the addon explorer if this is fixed.
https://github.com/MisterMakerNL/Freecad-Built-in-themes-beta

adrianinsaval, 2023-08-05:
> @adrianinsaval please transfer to flatpak repo, TIA

It looks like the problem comes from the stylesheet not really from flatpak.

adrianinsaval, 2023-08-05:
> https://github.com/FreeCAD/FreeCAD/blob/aa535962896a411e06f72a235ca58db0002907db/src/Gui/Stylesheets/Darker-blue.qss#L121

Can't we remove this line? I can't understand why we would need this. I can't think of a reason why you would want something so generic as QWidget to have a transparent background.

Rexbas, 2023-08-05:
>Can't we remove this line?

I tried it and removing this line also works.

MisterMakerNL, 2023-08-05:
Yeah, you are right not sure also why this is set to transparent!
Seems like the calltips was something else.

luzpaz, 2023-10-11:
bumping for status

MisterMakerNL, 2023-10-11:
I think this bug is dead, maybe OP can confirm it?

Rexbas, 2023-10-11:
This was fixed.

closed 2023-10-11 by Rexbas

schoenid, 2024-04-24:
Well this seems not to be fixed in 21.2.

I've got the exact same problem.
Unfortunately I can't add a screenshot, because the printscreen function of my OS is blocked by FreeCAD if a menu is open.
s.a https://github.com/FreeCAD/FreeCAD/issues/13614

[code]
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * pyrate (Disabled)
  * Glass
  * CadbaseLibrary 0.3.0
  * Defeaturing 1.2.2
  * Design456 0.0.1
  * ModernUI (Disabled)
  * pcb 6.2023.1
  * kicadStepUpMod 10.23.2
  * CurvedShapes 1.0.5
  * parts_library
  * Curves 0.6.24
  * Assembly4 0.50.9
  * freecad.gears 1.2.0
  * fasteners 0.5.13
  * frame 0.1.1
  * 3D_Printing_Tools
[/code]

schoenid, 2024-04-24:
Hmm...

It seems to work with no template or with ProDark ...
... but not with Darker-Blue, which is my favorite.

adrianinsaval, 2024-04-24:
I think the "fix" is was that those stylesheets where removed in 0.22. I'll see if I can remove that line I mentioned for the upcoming maintenance 0.21 release