FreeCAD/FreeCAD #10079: Toolbars disappeared after reopening an file and other

AlexanderGBerg, 2023-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33682 (Git)
Build type: Release
Branch: master
Hash: 9dce2525f8efe0873787f4388f10e11e168fe16b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

None

### Problem description

All toolbars are disappeared after reopening a file and have to be reactivated manually. Perferences will be ignored, after changing "switch workbench after loading "Start" to "Part Design" and activating "Close start page after loading".

### Anything else?

https://www.berg-kommunikation.de/test.FCStd

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-06:
Can you give clearer steps on how to reproduce this?

Syres916, 2023-08-11:
The nearest scenario to this that I've seen is when logging is enabled on the Windows shortcut, it's known to mess around with the UI.

maxwxyz, 2024-03-02:
@AlexanderGBerg is this still happening in the stable build? Could you try this with the newest weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
I cannot reproduce this behavior.

AlexanderGBerg, 2024-03-02:
I installed an older version of FreeCAD.

closed 2025-02-05 by github-actions[bot]