FreeCAD/FreeCAD #10080: [Problem]

tnozicka, 2023-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 41b058e)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

Arch

### Problem description

Solar diagram doesn't render correctly. It renders only the numbers and "SUMMER"/"WINTER" but all the lines are missing.
[image]

It can be reproduced even without the `Site` object:
```python
>>> import FreeCADGui, Arch
>>> Node = Arch.makeSolarDiagram(-46.38, -23.33, scale=10000, complete=True)
/app/lib/python3.10/site-packages/pysolar/solartime.py:111: UserWarning: I don't know about leap seconds after 2021
  warnings.warn \
>>> FreeCAD.Gui.ActiveDocument.ActiveView.getSceneGraph().addChild(Node)
>>> ### Begin command Std_ViewFitAll
>>> # Gui.SendMsgToActiveView("ViewFit")
>>> ### End command Std_ViewFitAll
```
[image]

The solar diagrams are expected to render lines, like here https://wiki.freecad.org/File:Freecad-solar-diagram.jpg

### Anything else?

```
$ flatpak info org.freecadweb.FreeCAD

FreeCAD - An open source parametric 3D CAD modeler

          ID: org.freecadweb.FreeCAD
         Ref: app/org.freecadweb.FreeCAD/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 0.21.0.33675
     License: LGPL-2.1
      Origin: flathub
  Collection: org.flathub.Stable
Installation: system
   Installed: 1,1 GB
     Runtime: org.kde.Platform/x86_64/5.15-22.08
         Sdk: org.kde.Sdk/x86_64/5.15-22.08

      Commit: 329e34750e26092a17667672a53f4a159aac3b3a3dab3cd415b12ec12e0a2ed4
      Parent: a88e131e0d2574a39d5862d1b23f657a3e96ba1b1a1bf45084ac29f319daa559
     Subject: Update to 0.21.0 release (817fc185)
        Date: 2023-08-03 21:15:14 +0000
```
AFAIK, this use pysolar 0.10
https://github.com/flathub/org.freecadweb.FreeCAD/blob/817fc18/org.freecadweb.FreeCAD.yaml#L672-L681

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-08-05:
Please give it a meaningful tile (now it is only _[Problem]_,surly not intented).

luzpaz, 2023-08-05:
Can someone reproduce this on anything beside flatpak ?

luzpaz, 2023-08-21:
@tnozicka can you reproduce on AppImage or Snap ?

tnozicka, 2023-08-23:
unfortunately, I don't have either to try (running Fedora Silverblue 38)

luzpaz, 2023-08-23:
@tnozicka please download the appropriate appimage (x86_64 or arch_64) from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and  report back

luzpaz, 2023-08-23:
@adrianinsaval do you mind transferring to FreeCAD flatpak repo ? TIA

adrianinsaval, 2023-08-24:
> @adrianinsaval do you mind transferring to FreeCAD flatpak repo ? TIA

That's not possible since that repo is in a different organization (flathub)

yorikvanhavre, 2023-09-01:
The same problem is in FreeCAD master code...
[image]
so it's a bug in FreeCAD ;) I'll look into it

luzpaz, 2023-09-04:
Found https://github.com/pingswept/pysolar/issues/139

bgbsww, 2023-10-30:
So, the issue here is the output of "    buf = shape.writeInventor(2,0.01)" around line 93 of ArchSite.py.

Earlier versions of freecad it looks like:
#Inventor V2.1 ascii 

Separator { 
}
#Inventor V2.1 ascii 

Separator { 
  Separator { 
    Material { diffuseColor 0 0 0} 
    DrawStyle { lineWidth 2 linePattern 0xffff } 
    Coordinate3 { 
      point [ 1250 0 0,
          1249.98 7.06925 0,
          1249.92 14.1383 0,
...

And current version it is:
#Inventor V2.1 ascii 

Separator { 
}
#Inventor V2.1 ascii 

Separator { 
  Separator {
    Material { diffuseColor 0 0 0}
    DrawStyle { lineWidth 2 linePattern 0xffff }
    Coordinate3 {
  point [
    1250 0 0
    1249.98 7.06925 0
    1249.92 14.1383 0
...

Note there aren't commas between the points.  This breaks the following parsing.

bgbsww, 2023-10-30:
However, it looks like is another potential bug in ArchRebar.py  which has very similar ( redundant? ) code that expects the comma to be there.

luzpaz, 2023-11-06:
ping @yorikvanhavre

luzpaz, 2023-11-08:
CC @amrit3701 @SurajDadral can y'all weigh in ?

yorikvanhavre, 2023-11-10:
@bgbsww your fix seems good to me... Mind to do a little PR? :innocent:

closed 2023-11-20 by yorikvanhavre

luzpaz, 2023-11-20:
Thanks @bgbsww !

PR #11339 (2023-11-20): Parse inventor format without commas between points
Fix #10080