FreeCAD/FreeCAD #10084: [Problem] Unnecessary Python varargs declaration

pieterhijma, 2023-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.0.33682 (Git)
Build type: DEBUG
Branch: master
Hash: 9dce2525f8efe0873787f4388f10e11e168fe16b
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * osh-autodoc-workbench
```

### Subproject(s) affected?

Core

### Problem description

As discussed in https://github.com/FreeCAD/FreeCAD/pull/10060#issuecomment-1664408907 and beyond, the file `Gui/View3DPy.cpp` contains several declarations of Python functions without arguments that are declared with `add_varargs_method`. This can be simplified with `add_noargs_method` removing the need to check the arguments.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-08:
@pieterhijma can this ticket be closed? 
Note: https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue

pieterhijma, 2023-11-08:
Yep, sorry, I'm more used to GitLab where "closes <issue>" are automatically added.

closed 2023-11-08 by pieterhijma