FreeCAD/FreeCAD #10086: When accesing the preferences it shows an error message

Alberto4343, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

When accesing the preferences it shows the error message i attached.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-06:
Can you try changing that path in the FEM preferences? The executable should exist in the V0.20 bin folder.

FEA-eng, 2023-08-06:
It's not a bug, it's just a reference to the CalculiX executable that was present in a folder with another version installed on the computer (and apparently removed). You have to change that reference manually or set "Search in known binary directories".

grubertechnic, 2023-08-06:
Also second that, that is no bug. The path must be adapted per hand, when the other binary is deleted.

Alberto4343, 2023-08-06:
Hi, I dont know well what is exactly a bug but i think it is a bug because it is an error in the development of freecad (in the path implemented in the source code) that causes a failure that causes that message to appear. If it was not a bug where do would i have to report it?

Alberto4343, 2023-08-08:
Hi, thank you for your answer. 
i didn't change that path manually, so that's not the cause of the problem.I'm trying in the 0.21 stable and it is not happening i will keep trying. The developer team should fix the problems that causes one to have this issue when uninstalling a version and installing a previous version.

grubertechnic, 2023-08-21:
I definitely see no issue her on FreeCADs site. If seems, that you do version jumping, there is no wonder, then you ruin your configs. Also we never had such an issue before and are not able to confirm, that you are not fiddle around in the preferences, since you seems fiddle around it according the other issues you opened.

@luzpaz, @Roy-043 or @FEA-eng : i think that is a candidate to close.

closed 2023-08-21 by luzpaz