FreeCAD/FreeCAD #10087: When i try to create a mesh via the FEM mesh from shape by Gmsh tool from a revolve created from a face it fails

Alberto4343, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

When i try to create a mesh via the FEM mesh from shape by Gmsh tool from a revolve created from a face it fails showing the following message:

22:25:15  Active analysis found: Analysis
22:25:18  Active analysis found: Analysis
22:25:18    ElementDimension: 3
22:25:18    Group meshing for analysis is set to true in FEM General Preferences. Are you really sure about this? You could run into trouble!
22:25:18    Empty reference: MaterialSolid
22:25:18    One material with no reference shapes. No need to make a group for materials.
22:25:18    C:\Users\Usuario\AppData\Local\Temp\fcfem_lqfxygf3\Body_Geometry.brep
22:25:18    C:\Users\Usuario\AppData\Local\Temp\fcfem_lqfxygf3\Body_Mesh.unv
22:25:18    C:\Users\Usuario\AppData\Local\Temp\fcfem_lqfxygf3\shape2mesh.geo
22:25:18    C:/Program Files/FreeCAD 0.20/bin/gmsh.exe
22:25:18    New mesh was added to the mesh object.
22:25:18  Gmsh had warnings ...
22:25:18  Error   : PLC Error:  A segment and a facet intersect at point
Error   : Invalid boundary mesh (segment-facet intersection) on surface 23, intersection (-91.0614,81.5289,-80.527)
Error   : No elements in volume 1 
Error   : ------------------------------
Error   : Mesh generation error summary
Error   :     0 warnings
Error   :     3 errors
Error   : Check the full log for details
Error   : ------------------------------

### Anything else?

File:
[Nuevo aparato.zip](https://github.com/FreeCAD/FreeCAD/files/12268664/Nuevo.aparato.zip)

Image of the solid generated:
[image]
 
 Images of the face that i used to generate the solid:
[image]
[image]

Images of the sketch that i used to generate the face:
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-06:
This part is highly thin-walled and thus not so easy to mesh. You have to carefully choose the mesh size. Try with 10 mm, for example. It would be best to model this as a surface and mesh with shell elements.

Alberto4343, 2023-08-06:
i tried that and it worked but it consumes so many resources from the laptop im using. i will investigate your solution. it would be nice that the FEM mesh from shape by Gmsh tool do the necesary calculations to create smaller elements where needed automatically.

luzpaz, 2023-08-06:
Verdict? (keep open or close)

Alberto4343, 2023-08-07:
it is solved. i will close it. thank you so much.

Alberto4343, 2023-08-07:
Ok i don't know how to close it, i dont know how to use this, it is my first time. maybe my type of user can't and you do it.

closed 2023-08-07 by Alberto4343