FreeCAD/FreeCAD #10088: [Problem] Window layout redraw is botched

DubDublin, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Window layout gets botched and is apparently unfixable w/o restarting FreeCAD, and sometimes, Windows.  (I'm now using 0.21, but this was a problem in 0.20.2 as well...) There seem to be several (probably related) problems here:  1) After either the monitor going to sleep, or the laptop going to sleep and waking again, FreeCAD does not correctly redraw the screen - all menus, controls, etc. are outside the viewport, and resizing, minimizing, maximizing, etc. the window does not provoke a refresh as expected.  See attached image.  The only way I've found to fix this is kill the app and restart it.  (Sometimes a Windows restart is required?)   2)  FreeCAD always wants to be on the primary display, which is almost always the wrong one for a laptop with an attached monitor.  Simply moving the FreeCAD window to the secondary/external display (by either Shift-Win-RtArrow or draging and then maximizing) can also cause this problem. 3) More troublingly, it appears that in some cases, the controls may actually still be there, but are not visible - you can roll over the top area of the screen and see tooltips, but can't see the controls themselves.  This also makes using the menus via the Alt-key and mouse impossible - I had to file this bug report via Alt-H, then use the keyboard ONLY to navigate down to "Report a bug", since using the mouse to try to select one of the menu items blows you out and does God only knows what...
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DubDublin, 2023-08-06:
BTW, problem is worse when using FreeCAD on second monitor, but I'm pretty sure I've seen it on the primary monitor as well.

xtemp09, 2023-08-06:
This problem could be vendor-dependent. I use few displays at once and have never encounter this. I use Intel GPU and it's not a laptop, so I don't have energy-saving settings.

Could you also report your CPU, GPU, their driver versions and, perhaps, what settings are on (`Settings  > System  > Power & sleep  > Additional power settings`)?

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

DubDublin, 2023-08-06:
Sure: Here you go - nothing very exotic at all (This is a newish laptop, and I haven't even gotten a dock for it yet - it's just the monitor plugged directly into the laptop.  It's a Lenovo 9i 14" i7, with a Dell P2815Q 4K monitor connected via HDMI) - there is nothing else attached - until I get a dock, I'm using the laptop's built-in keyboard and a Microsoft Precision bluetooth mouse as the only peripheral other than the monitor.)

[image]

DeviceName                                                 Manufacturer              DriverVersion
----------                                                 ------------              -------------
Local Print Queue                                          OKI                       10.0.22621.1
Local Print Queue                                          Microsoft                 10.0.22621.1
Local Print Queue                                          Microsoft                 10.0.22621.1
Local Print Queue                                          Microsoft                 10.0.22621.1
Local Print Queue                                          Microsoft                 10.0.22621.1
Generic software device                                    Microsoft                 10.0.22621.1
Generic software device                                    Microsoft                 10.0.22621.1
Generic software device                                    Microsoft                 10.0.22621.1
Generic software device                                    Microsoft                 10.0.22621.1
Generic software device                                    Microsoft                 10.0.22621.1
Generic software device                                    Microsoft                 10.0.22621.1
Generic software device                                    Microsoft                 10.0.22621.1
Computer Device                                            Microsoft                 10.0.22621.1
Remote Desktop Device Redirector Bus                       Microsoft                 10.0.22621.608
Universal Device Client Device                             Lenovo                    23.5.0.38
System Interface Foundation V2 Device                      Lenovo                    1.2.0.11
Plug and Play Software Device Enumerator                   (Standard system devices) 10.0.22621.1
Microsoft System Management BIOS Driver                    (Standard system devices) 10.0.22621.1
NDIS Virtual Network Adapter Enumerator                    Microsoft                 10.0.22621.1
Microsoft Hyper-V Virtual Disk Server                      Microsoft                 10.0.22621.1702
Microsoft Basic Render Driver                              Microsoft                 10.0.22621.608
Microsoft Hyper-V PCI Server                               Microsoft                 10.0.22621.1
System Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Device Firmware                                            Microsoft                 10.0.22621.1
Microsoft UEFI-Compliant System                            Microsoft                 10.0.22621.1
ACPI Thermal Zone                                          (Standard system devices) 10.0.22621.1
ACPI Fan                                                   (Standard system devices) 10.0.22621.1
ACPI Fan                                                   (Standard system devices) 10.0.22621.1
ACPI Fan                                                   (Standard system devices) 10.0.22621.1
ACPI Fan                                                   (Standard system devices) 10.0.22621.1
ACPI Fan                                                   (Standard system devices) 10.0.22621.1
Motherboard resources                                      (Standard system devices) 10.0.22621.1
UCM-UCSI ACPI Device                                       Microsoft                 10.0.22621.1
Trusted Platform Module 2.0                                (Standard)                10.0.22621.608
Intel(R) Innovation Platform Framework Manager             Intel                     1.0.11000.27915
HID-compliant system controller                            (Standard system devices) 10.0.22621.819
HID-compliant consumer control device                      Microsoft                 10.0.22621.1
HID Keyboard Device                                        (Standard keyboards)      10.0.22621.1
Converted Portable Device Control device                   Microsoft                 10.0.22621.1
Portable Device Control device                             Microsoft                 10.0.22621.1
HID-compliant vendor-defined device                        (Standard system devices) 10.0.22621.819
[... cut ...]
Volume Manager                                             Microsoft                 10.0.22621.608

                                                                                     2:10.0,2:6.3...
                                                                                     2:10.0,2:6.3...
                                                                                     2:10.0,2:6.3...
                                                                                     2:10.0,2:6.3...
[image]

xtemp09, 2023-08-07:
Could you create a boolean parameter named _DisableDpiScaling_ in `Tools → Edit parameters` at `BaseApp/Preferences/HighDPI` and set it to _true_? Or _false_ and see how it influences.

DubDublin, 2023-08-08:
OK, will do - just woke it up again after being AFK for a while, here's what I saw that may help:  when the computer woke back up, it momentarily flashed the FreeCAD window on the laptop's screen before restoring it to the 4K external display.  (This is normal, if awkward, wake behavior for Win11 and this laptop/monitor combo - my old Win10/Surface Pro did not do this even on the same monitor.)  Here's what's interesting:  The FreeCAD window was correctly laid out with all the controls when it momentarily flashed onto the laptop screen (as though it was maximized there), but was then instantly botched again as in the original image I posted when it showed up on the external 4K screen.  There is no way I can find to force a redraw/relayout short of killing the app and starting over.

DubDublin, 2023-08-08:
With this set to True, FreeCAD starts up with all the UI elements being *very* small, with tiny icons and unreadable text.  (That may make sense, since both the built-in display and the external display have high pixel counts - 3840x2400 for the built-in display, and of course the usual 4K 3840x2160 for the external monitor.) 

It did NOT botch the redraw after my first try with this setting.

I'm going to try again a couple of times with True before setting it to False to see if that changes anything...

luzpaz, 2023-08-11:
@DubDublin thanks! keep us posted!

DubDublin, 2023-08-14:
UI is of course, unusable with _DisableDpiScaling_  set to True, so it's hard to do much to test how it works.  I haven't seen it fail the same way, but I'm not sure it's a good test.  Problem does persist with it set to False, just as it does when it's not set at all, like it is out of the box...

DubDublin, 2023-08-31:
Just documenting that this problem still persists, regardless of anything I've tried.  There is a bug with redraw after waking that is different than just resizing the window when moving to a new monitor - large parts of the UI are off-screen and cannot be brought back.  So far, the only remedy is still to close FreeCAD and restart it.

luzpaz, 2024-08-25:
@DubDublin can you still reproduce this on latest development version ? 
(https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

maxwxyz, 2024-10-06:
any news on this, is this still relevant? I'll close if there is no feedback.

maxwxyz, 2024-11-01:
closing for no feedback

closed 2024-11-01 by maxwxyz

andylovescode, 2025-03-10:
This issue is still a problem, even on the latest version!