FreeCAD/FreeCAD #10089: DImensions in the Property dialoge not displayed when entering a value

Panda631, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * EM
  * fasteners 0.4.56
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * sheetmetal 0.2.61
[/code]
```

### Subproject(s) affected?

None

### Problem description

For example, start creating a wedge.
All dimensions can be entered in the input dialog. If the dimensions (e.g. mm) are overwritten, they are no longer displayed. This was not the case in version v0.20.2. There the dimensions were added automatically again.
After closing and reopening the dialog, the dimensions are displayed correctly again.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-06:
Confirmed.

Syres916, 2023-08-06:
I'm trying to replicate this on Linux and getting nowhere. In order to eliminate a packaging issue, @Panda631 could you please start the simple procedure from a clean start of FreeCAD 0.21 up to the point that the incorrectly displayed constraint is shown and copy all the contents from the Python Console into a reply here please. I can then compare to my Linux output to hopefully help identify the root cause.

In the 0.20.2 release cycle, a very strange angle constraint issue occurred and was found to only apply to the Windows LibPack build which was fixed by repackaging but I'm still not clear what the actual root cause was.

Syres916, 2023-08-06:
Ignore the above, I've now replicated what the OP is reporting.

Last Good Build:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.30377 (Git) AppImage   15th Sept 2022
Build type: Release
Branch: master
Hash: 2462f4c7a05745488bb9e81ccc7ba0f386f78849
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
```

First Bad Build:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31709 (Git) AppImage   30th Jan 2023
Build type: Release
Branch: master
Hash: e188802ca6997d2564e7570ab648462e6a059f87
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
```

I'll track down the commit and report back.

Syres916, 2023-08-06:
I've had to abandon this, uncompilable commits are the bane of triaging.

grubertechnic, 2023-08-06:
Thought, this was intended, because of the imperial units parsing. (And also do not like that). At least this behavior was introduced one of the better detection (with divide signs, " and ' units) imperial units commits.

Syres916, 2023-08-07:
@ConnorWorrell would you care to advise if the issue reported is by design or could a fix be put in place?

ConnorWorrell, 2023-08-08:
This was intentional, under the hood it's still tracking/adding units as needed, it no longer updates the display.

I agree anything where units are being added to the end should update, but anything more complicated than that should not.

grubertechnic, 2023-08-08:
To be honest, that this behavior, that the reporter reported, annoys me also. So if the units are showing again, it would be really nice.

luzpaz, 2023-08-16:
Updated the ticket to indicate this is now a discussion about presenting Pros and Cons for displaying units

grubertechnic, 2023-08-21:
Where are cons when units showed? Ambiguity is always bad, unambiguousness is always good.

luzpaz, 2023-09-24:
bumping

FEA-eng, 2026-10-03:
I still don't know which dimension input boxes are discussed here.