FreeCAD/FreeCAD #10090: [Problem] FreeCAD doesn't start on MacOS Catalina 10.15.7

X-Ryl669, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
Can't get to that point
```

### Subproject(s) affected?

None

### Problem description

Checked #9465 and maybe it's related, but it's not exactly the same issue. I've downloaded FreeCAD DMG for intel on 08/05/23, and it doesn't start. 

When launched from the terminal, it gives this output:
```bash
$ MacOS/FreeCAD

LIST OF PACKAGES:
#
# Name                    Version                   Build  Channel
aiohttp                   3.8.5           py310h6729b98_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
arpack                    3.7.0                hefb7bc6_2    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
blas                      2.117                  openblas    conda-forge
blas-devel                3.9.0           17_osx64_openblas    conda-forge
blinker                   1.6.2              pyhd8ed1ab_0    conda-forge
boost                     1.78.0          py310h3e792ce_4    conda-forge
boost-cpp                 1.78.0               h31500c2_2    conda-forge
brotli                    1.0.9                hb7f2c08_9    conda-forge
brotli-bin                1.0.9                hb7f2c08_9    conda-forge
brotli-python             1.0.9           py310h7a76584_9    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.19.1               h0dc2134_0    conda-forge
ca-certificates           2023.7.22            h8857fd0_0    conda-forge
cairo                     1.16.0            h904041c_1014    conda-forge
calculix                  2.20                 h61d0cde_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cgal-cpp                  5.5.2                hd6e141a_0    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
coin3d                    4.0.0             h5af9841_1012    conda-forge
contourpy                 1.1.0           py310h88cfcbd_0    conda-forge
curl                      8.1.2                hbee3ae8_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
docutils                  0.20.1          py310h2ec42d9_0    conda-forge
double-conversion         3.2.0                hf0c8a7f_1    conda-forge
eigen                     3.4.0                h1c7c39f_0    conda-forge
expat                     2.5.0                hf0c8a7f_1    conda-forge
ffmpeg                    5.1.2           gpl_h8b4fe81_106    conda-forge
flann                     1.9.1             h56de9e4_1011    conda-forge
fltk                      1.3.8                h7ffa276_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h5bb23bf_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.41.1          py310h6729b98_0    conda-forge
freecad                   0.21.0          py310hfdc9f54_0    freecad/label/stable
freeimage                 3.18.0              h4fbe5d5_12    conda-forge
freetype                  2.12.1               h3f81eb7_1    conda-forge
[... 184 lines cut ...]
xorg-libxfixes            5.0.3             h0d85af4_1004    conda-forge
xorg-libxrender           0.9.10            h0d85af4_1003    conda-forge
xorg-renderproto          0.11.1            h0d85af4_1002    conda-forge
xorg-xextproto            7.3.0             hb7f2c08_1003    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.6                h775f41a_0    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
yarl                      1.9.2           py310h6729b98_0    conda-forge
zlib                      1.2.13               h8a1eda9_5    conda-forge
zstd                      1.5.2                h829000d_7    conda-forge
dyld: Library not loaded: @rpath/libCoin.80.dylib
  Referenced from: /Applications/FreeCAD.app/Contents/Resources/lib/libFreeCADGui.dylib
  Reason: no suitable image found.  Did find:
	/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib: code signing blocked mmap() of '/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib'
	/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib: stat() failed with errno=1
	/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib: code signing blocked mmap() of '/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib'
	/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib: stat() failed with errno=1
MacOS/FreeCAD: line 19:  1155 Abort trap: 6           "${PREFIX}/bin/freecad" $@
```

So it seems to fail on code signing for libCoin.80.dylib.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-08-07:
@chennes could you explain what process you followed to replace the library and then signing the bundle?

chennes, 2023-08-07:
I just replaced the single library file (not the links) and reran the entire signing process. I will replace the links next time.

luzpaz, 2023-08-11:
@chennes verdict on this ticket ?

chennes, 2023-08-11:
None yet: I don't have access to this OS so testing possible solutions is slow. Apple has changed their code signing requirements significantly in recent OSes, and it appears that what works in new versions does not necessarily work in old ones.

nonno-cicala, 2023-08-12:
Hi, I just installed FreeCAD-0.21.0-coin_patch-mac-intel_x86.dmg on Mojave 10.14.6 but I still get the same error as OP ☹️

czechcore, 2023-08-16:
Same here
macOS 11.7.6.
FreeCAD-0.21.0-coin_patch-mac-intel_x86.dmg

jmanley, 2023-08-16:
I seem to have the same problem on OS 11.6.1 using the downloaded version FreeCAD-0.21.0-mac-intel_x86.dmg from today.

```
Dyld Error Message:
  dyld: Using shared cache: 80896B32-9EB1-3A10-A7C5-449B15305DE2
Library not loaded: @rpath/libCoin.80.dylib
  Referenced from: /Applications/FreeCAD.app/Contents/Resources/lib/libFreeCADGui.dylib
  Reason: no suitable image found.  Did find:
	/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib: code signing blocked mmap() of '/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib'
	/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib: code signing blocked mmap() of '/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib'
	/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib: stat() failed with errno=1

```

chennes, 2023-08-16:
I suspect that anything prior to Ventura (Mac OS 13) is going to have an issue here: these libraries were created using the Mac OS 10.9 SDK, which apparently used different signing technology. For whatever reason Ventura seems happy to launch the executable anyway, but the older OSes are less tolerant. I'm looking into whether the best path forward is to try to recompile the whole chain with a newer SDK, or to just release an unsigned version for those on the older OSes.

chennes, 2023-08-16:
I've uploaded an unsigned version of the DMG that should work on versions of MacOS prior to Mac OS 13 (Ventura) -- I'd appreciate your feedback on whether this works on your systems. 
https://github.com/FreeCAD/FreeCAD/releases/download/0.21.0/FreeCAD-0.21.0-mac-intel_x86-unsigned.dmg

rainell, 2023-08-16:
@chennes FreeCAD-0.21.0-mac-intel x86-unsigned.dmg works on macOS 10.15.7 (Catalina). Thank you.

nonno-cicala, 2023-08-16:
@chennes still fails for me on 10.14.6 (Mojave) but with a new error, if I try to run it from the command line I get at the end:
```
dyld: Library not loaded: @rpath/libCoin.80.dylib
  Referenced from: /Applications/FreeCAD.app/Contents/Resources/lib/libFreeCADGui.dylib
  Reason: no suitable image found.  Did find:
	/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib: cannot load 'libCoin.80.dylib' (load command 0x80000034 is unknown)
	/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib: stat() failed with errno=1
	/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib: cannot load 'libCoin.80.dylib' (load command 0x80000034 is unknown)
	/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib: stat() failed with errno=1
	/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib: cannot load 'libCoin.80.dylib' (load command 0x80000034 is unknown)
	/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib: stat() failed with errno=1
/Applications/FreeCAD.app/Contents/MacOS/FreeCAD: line 19: 44937 Abort trap: 6           "${PREFIX}/bin/freecad" $@
```

chennes, 2023-08-17:
@nonno-cicala can you try the Conda bundle from here? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.21.0

X-Ryl669, 2023-08-17:
It also works for me on 10.15.7 (with the usual unsigned application message that needs Ctrl + open to load).
@nonno-cicala The error is not the same. It seems that the symbolic link named "libCoin.80.dylib" is missing or not functional in your version. If you can't find it, it looks like this:
```bash
% ls -l /Applications/FreeCAD.app/Contents/Resources/lib/libCoin.*
-rwxrwxrwx@ 1 user  admin  7981056 16 aoû 21:19 /Applications/FreeCAD.app/Contents/Resources/lib/libCoin.4.0.0.dylib
lrwxr-xr-x  1 user  admin       19 16 aoû 22:14 /Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib -> libCoin.4.0.0.dylib
lrwxr-xr-x  1 user  admin       19 16 aoû 22:14 /Applications/FreeCAD.app/Contents/Resources/lib/libCoin.dylib -> libCoin.4.0.0.dylib
```

thethinkingman, 2023-08-17:
I had the dyld: Library not loaded: @rpath/libCoin.80.dylib error but I have downloaded and run the unsigned version (above) and FreeCAD starts up. I started it using both from Finder as well as Terminal (/Applications/FreeCAD.app/Contents/Resources/bin/)

I am running Big Sur (11.7.8)

twbaumann, 2023-08-17:
Just to add my findings to the list, I am on **Mac OS X Mojave 10.14.6** (18G9323) because I have an old but trusty MacPro5,1. I tried four 0.21.0 release files and for me the only one that works is [FreeCAD_0.21.0-2023-08-02-conda-macOS-x86_64-py310.dmg](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/0.21.0/FreeCAD_0.21.0-2023-08-02-conda-macOS-x86_64-py310.dmg).

The other three files I downloaded and tried yesterday made FreeCAD crash on startup with SIGABRT and termination reason 'DYLD, [0x1] Library missing' and 'Library not loaded: @rpath/libCoin.80.dylib':

[FreeCAD-0.21.0-mac-intel_x86.dmg](https://github.com/FreeCAD/FreeCAD/releases/download/0.21.0/FreeCAD-0.21.0-mac-intel_x86.dmg) (official link)
_code signing blocked mmap() of '/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib'_
_code signing blocked mmap() of '/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib'_

[FreeCAD-0.21.0-coin_patch-mac-intel_x86.dmg](https://github.com/FreeCAD/FreeCAD/releases/download/0.21.0/FreeCAD-0.21.0-coin_patch-mac-intel_x86.dmg)
same as above

[FreeCAD-0.21.0-mac-intel_x86-unsigned.dmg](https://github.com/FreeCAD/FreeCAD/releases/download/0.21.0/FreeCAD-0.21.0-mac-intel_x86-unsigned.dmg)
_/Applications/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib: cannot load 'libCoin.80.dylib' (load command 0x80000034 is unknown)_
_/Applications/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib: cannot load 'libCoin.80.dylib' (load command 0x80000034 is unknown)_

chennes, 2023-08-17:
Thanks for the thorough report @twbaumann -- so that tells me that replacing the Coin library on an _ad hoc_ basis is failing on certain OSes, but is working on others. @adrianinsaval the upshot is that we really need to figure out how to get that older Coin build into the Conda. Any ideas on that front?

adrianinsaval, 2023-08-17:
the report however says that the weekly build is the one that's working, and that uses the regular coin package... maybe openbrain's PR did fix the issue but since it was compounded with the signing problem we thought it didn't?

I think we should try building 0.21 with that PR as a patch and not signed.

Aleksandr-ru, 2023-08-17:
> I've uploaded an unsigned version of the DMG that should work on versions of MacOS prior to Mac OS 13 (Ventura) -- I'd appreciate your feedback on whether this works on your systems. https://github.com/FreeCAD/FreeCAD/releases/download/0.21.0/FreeCAD-0.21.0-mac-intel_x86-unsigned.dmg

Works fine at MacOS 10.15.7
Thank You!

nonno-cicala, 2023-08-17:
> @nonno-cicala can you try the Conda bundle from here? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.21.0

@chennes, I tried the same Conda bundle that worked for @twbaumann and it worked for me too on 10.14.6! 🎉 
[FreeCAD_0.21.0-2023-08-02-conda-macOS-x86_64-py310.dmg](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/0.21.0/FreeCAD_0.21.0-2023-08-02-conda-macOS-x86_64-py310.dmg) 👈 🆗 

>  @nonno-cicala The error is not the same.

@X-Ryl669, it was the same error with the standard package and the coin_patched one, it changed with the unsigned package

Thanks everyone!

BPmnstuff, 2023-09-06:
I have also verified the unsigned version works on my installation of Catalina 15.7.  Thank you

adrianinsaval, 2023-10-09:
what's the status on this for the 0.21.1 release? which bundle works and which doesn't?

adrianinsaval, 2024-01-15:
I believe this has been solved, feel free to reopen if it's still a problem

closed 2024-01-15 by adrianinsaval