FreeCAD/FreeCAD #10092: [Problem] Allow user to merge (apply coincidence constraint) closely located points

raindropsfromsky, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33658 (Git)
Build type: Release
Branch: master
Hash: ba5f7e6d75dee09ec7076ad24895b04e56f21b63
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/India (en_IN)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.60n
  * ArchTextures
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * btl
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * ExplodedAssembly
  * fasteners 0.4.59
  * freecad.gears 1.0.0
  * Glass (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * SearchBar 1.0.2
[/code]
```

### Subproject(s) affected?

None

### Problem description

Often people cannot pad a sketch because [the wire is not closed](https://youtu.be/g8EhDY9cqe8?t=416). Often the reason is that two points that _appear_ to be connected through a coincident constraint are actually not connected.

It is tedious to find those disconnected points and connect those points manually.

Instead, FC should allow us to rapidly connect such points semi-automatically through a "**Merge**" command, as in Blender.
Blender has a command to merge any points that are within a user-specified distance.
In FC, just apply a coincident constraint to those points.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-07:
Such functionality is available as Validate sketch --> Missing coincidences. A tool to do it directly in the sketch would also be nice but for now you have to use the Validate sketch feature.

maxwxyz, 2024-03-09:
This tool (merge with distance tolerance) as well as the other sketch validation tools should be all available with an active sketch. Currently when in an active Sketch in the Sketcher WB if you exit it it goes back to the PartDesign WB (wanted behavior) but then I have to manually select the sketch, change to the Sketcher WB, click validate sketch button and after fixed manually return to the PartDesign WB. 
It is also very hidden from the user as this issue demonstrates. 
@PaddleStroke would it be possible to include the `Sketcher validation`to work in an active sketch? Currently its also using the task panel.

Maybe these features could also be included in the right click menu when selecting a sketch in the model tree.

PaddleStroke, 2024-03-12:
Same issue as https://github.com/FreeCAD/FreeCAD/issues/11799 I guess.
Possible yes surely. Easy? That's another question.
Imo blue points and red points would be easier to do