FreeCAD/FreeCAD #10093: Mesh boundary layer have no effect in the FEM mesh

Alberto4343, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

When i applied the Mesh boundary layer it has no effect in the FEM mesh the elements keep being the same size.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-06:
Please reproduce in latest stable (v0.21)

Alberto4343, 2023-08-06:
Ok, im going.

FEA-eng, 2023-08-07:
But you know what boundary layer is for ? It creates sort of offset layers in the mesh to properly resolve the near-wall flow. There's  usually no need to use this in structural FEM.

Alberto4343, 2023-08-08:
@luzpaz i tried in the stable v0.21.0 version and it still happens.

@FEA-eng i was trying to use it to reduce the element count and to improve the analysis of a fluid.

Here is a file where it happens.

[Nuevo aparato.zip](https://github.com/FreeCAD/FreeCAD/files/12286068/Nuevo.aparato.zip)

grubertechnic, 2023-08-08:
@luzpaz or @FEA-eng : can we close this or #10096 ticket, both is about the same topic and file. No need to be redundant.

Alberto4343, 2023-08-08:
@grubertechnic, hi, this is about boundaryLayer, the other one is about MeshRegion.

Alberto4343, 2023-08-08:
@luzpaz:

Do you refer to this?:

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
Installed mods: 
  * Help 1.0.3
[/code]

FEA-eng, 2023-08-13:
@Alberto4343 Are you sure this fluid domain is correctly defined ? What kind of device is this? How should the fluid flow through it?

luzpaz, 2023-08-13:
> It would be best to discuss it on the forum though.

@Alberto4343 can't seem to log in to the forum for some reason.

luzpaz, 2023-11-08:
What's the status of this ticket ?

FEA-eng, 2024-02-18:
@luzpaz or @maxwxyz I think that this issue can be closed as there's neither proof of any bug nor feedback from the user.

maxwxyz, 2024-02-19:
Cosed for no feedback.

closed 2024-02-19 by maxwxyz