FreeCAD/FreeCAD #10094: bug:TechDraw_AxoLengthDimension is out of infinity

3xxx, 2023-08-06:
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33719 (Git)
Build type: Release
Branch: master
Hash: 2a88c7c7dfed699d2a4944bda13bf8626ed39315
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Chinese/China (zh_CN)
Installed mods: 
  * A2plus 0.4.60h
  * Assembly3 0.11.4
  * Assembly4 0.12.5
[/code]

[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
Installed mods: 
  * A2plus 0.4.60h
  * Assembly3 0.11.4
  * Assembly4 0.12.5
[/code]
In the techdraw workbench, the annotation lines of the isometric side diagram run out of infinity, resulting in the inability to select and adjust, which may be the size of my model is relatively large, with a length, width and height of 30m
[image]

luzpaz, 2023-08-06:
Thanks @3xxx. Note: the correct repository to file bugs is FreeCAD/FreeCAD (the one I transferred the issue to). 

Can you zip an FCStd file and drag-drop it in to this ticket for us to recreate the bug please? TIA

3xxx, 2023-08-07:
[厂房-旧.zip](https://github.com/FreeCAD/FreeCAD/files/12279040/-.zip)
Thank you.

luzpaz, 2023-08-16:
@benj5378 do you mind weighing-in ?

WandererFan, 2023-08-16:
More of an issue for @edi271 I think.

luzpaz, 2023-09-09:
CC @edi271

luzpaz, 2023-11-01:
@pavltom perhaps you can take a look at this issue ?

edi271, 2023-11-02:
can't see any context between this issue and the github account of pavltom.

The AxoLengthDimension tool is a short Python script (scr/Mod/TechDraw/CommandAxoLengthDimension.py).

Without test file I can't do anything.

pavltom, 2023-11-02:
Hi, probably @luzpaz was trying to find anybody, who could help.

I have looked in the 3xxx's document and it seems the dimension itself is ok, only placed too far away (X = -11m, Y = 20m). If set to [0, 0], everything looks and works as expected. From my point of view this is rather incovenience (dimension automatically placed too far), than a fundamental issue. In any case, a fix would be nice.

luzpaz, 2023-11-02:
> Hi, probably @luzpaz was trying to find anybody, who could help.

Correct :+1: 

Thanks for weighing in

pavltom, 2023-11-06:
The PR above should resolve the issue. The existing dimensions too far away need the dimension X and Y coordinates set manually via Data property editor, the newly created ones should be placed as expected.

closed 2023-11-06 by WandererFan

luzpaz, 2023-11-07:
Thank you @pavltom :pray:

PR #11293 (2023-11-06): [TechDraw] Fix issue #10094
Resolves #10094