FreeCAD/FreeCAD #10095: Clipping plane is too large and doesn´t permit to hide it parcially when in the the left sidebar and in the bottom sidebar

Alberto4343, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

Clipping plane is too large and doesn´t permit to hide it parcially when in the the left sidebar or in the bottom sidebar. It consumes too much space.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-06:
Do you mind providing a screenshot please?

Alberto4343, 2023-08-06:
Note: I don't have a good english level.
Yes. 

This is the minimum width when in left sidebar:
[image]

This is the minimum height when in bottom sidebar:
[image]

luzpaz, 2023-10-25:
Related to #8477  
It looks like a duplicate

closed 2023-11-08 by luzpaz