FreeCAD/FreeCAD #10096: Mesh Region doesn´t act as expected

Alberto4343, 2023-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

Mesh Region doesn´t act as expected, it extends so much more than what is described in the wiki.

### Anything else?

### Mesh Region with a Max element size of 4mm:

[image]

### Wiki:

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-07:
To which geometric entities did you assign this mesh region ? Can you share the file with it ?

Alberto4343, 2023-08-07:
I created another case of this.

I assigned the Mesh regions to the curved faces, the extarnal curved face and the internal curved face.

This is the file:

[Nuevo aparato.zip](https://github.com/FreeCAD/FreeCAD/files/12285741/Nuevo.aparato.zip)

grubertechnic, 2023-08-08:
3 things:

0.: the model can be created much easier, it is overcomplicated done, but this have nothing to do with FEM

1.: You have a wall thickness ~1mm. Also you have net the overall mesh option to automatic, so it will be fine anyway.

2.: mesh region is handled generally from gmsh a little bit odd, but it is also written in the wiki, see https://wiki.freecad.org/FEM_MeshGmshFromShape#Mesh_Growth.

If you have specify 2 regions (example faces) and there is a region (example a face) between, then the growing transition is between these two faces is done in the middle. Since your example is circular topology*, the face in the middle gets no transition, because in "both sides" the mesh region value are the same.

* (not 100% since it is a b-spline, because it is not made well, see point 0., but for now it does not matter),

Alberto4343, 2023-08-08:
Note: i don't have a good level of english.

Does the Freecad team develop Gmsh? If it is the case they should repair that behavior if not the they should find another mesh generator or build one.
What you explained is the workaround i'm using and it is working nicely.
I'm sorry i don't understand well what markdown automatism is, it appears to refer to my mode of writting but i'm not sure.

FEA-eng, 2023-08-09:
> Does the Freecad team develop Gmsh? If it is the case they should repair that behavior if not the they should find another mesh generator or build one.

No, Gmsh is developed by a separate team. You can raise a request on their GitLab, I'm sure they will be pleased: https://gitlab.onelab.info/gmsh/gmsh/-/issues

Alberto4343, 2023-08-09:
I will, thank you.

luzpaz, 2023-08-16:
@Alberto4343 can you post the link to the ticket or discussion you started with gmsh team ?

FEA-eng, 2024-02-18:
@luzpaz or @maxwxyz I think that this issue can be closed as there's neither proof of any bug nor feedback from the user.

maxwxyz, 2024-02-19:
Cosed for no feedback.

closed 2024-02-19 by maxwxyz