FreeCAD/FreeCAD #10098: Issue with working planes (Draft_SelectPlane in gui_selectplane.py)

keinze, 2023-08-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * BIM 2021.12.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Arch

### Problem description

Suddenly alingiong the working plane to a surface failed. If I select any surface there is the following trace:

```
01:47:27  Running the Python command 'Draft_SelectPlane' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.20\Mod\Draft\draftguitools\gui_selectplane.py", line 138, in Activated
    if self.handle():
  File "C:\Program Files\FreeCAD 0.20\Mod\Draft\draftguitools\gui_selectplane.py", line 218, in handle
    FreeCAD.DraftWorkingPlane.alignToFace(sel.SubObjects[0],
  File "C:\Program Files\FreeCAD 0.20\Mod\Draft\WorkingPlane.py", line 579, in alignToFace
    if shape.ShapeType == 'Face':

cannot determine type of null shapeObject with null Part.Shape geometry:'Site'
01:47:37  Wähle eine Fläche, 3 Punkte oder einen WP Proxy um die Zeichenebene zu definieren
01:47:53  Running the Python command 'Draft_SelectPlane' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.20\Mod\Draft\draftguitools\gui_selectplane.py", line 138, in Activated
    if self.handle():
  File "C:\Program Files\FreeCAD 0.20\Mod\Draft\draftguitools\gui_selectplane.py", line 218, in handle
    FreeCAD.DraftWorkingPlane.alignToFace(sel.SubObjects[0],
  File "C:\Program Files\FreeCAD 0.20\Mod\Draft\WorkingPlane.py", line 579, in alignToFace
    if shape.ShapeType == 'Face':

cannot determine type of null shape
```
I can only set the working plane to standard XY, XZ, YZ axis if no object is selected. 
Hoever the working plane has strange behaviour it is no longer through the center, but at a far random distance 

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-07:
Do you mind trying to reproduce this on latest stable (v0.21) ?

yorikvanhavre, 2023-08-07:
I confirm aligning the WP to a face works normally. Here the problem is this:

> cannot determine type of null shape

If you say you have a site selected, that's a bug in 0.20: The site object, even having no shape, gathers the 3D representation of its subobjects, so it appears to have an own shape and is selectable, although it's actually empty.

This has been fixed in 0.21 already. So there is no fix needed here, just head to the releases page here on github and download it!

keinze, 2023-08-07:
I can confirm, selection problem is solved in release 0.21.