FreeCAD/FreeCAD #10099: [Problem] Open file dialog freezes on windows (upstream bug solved in Qt6)

luzpaz, 2023-08-07:
This ticket is a placeholder for #5853 and depends on #6992

luzpaz, 2023-08-11:
In short when Qt6 for windows port is implemented, this ticket should be closed.

adrianinsaval, 2023-08-22:
Please add a proper description in this issue itself, our guidelines says that an issue should be understood without requiring to follow links

adrianinsaval, 2023-08-22:
Btw I didn't see any confirmation that this issue is actually solved in qt6, just speculation that it might be

chennes, 2023-08-22:
I'm hoping to have the Qt6 LibPack ready in a couple of weeks, so we should be able to find out for sure soon.

luzpaz, 2024-10-20:
@chennes can this ticket be closed ?

chennes, 2024-10-20:
> In short when Qt6 for windows port is implemented, this ticket should be closed.

If this is true, then yes. I don't know anything about this issue, and don't experience any problems with Qt's open file dialog in Qt6 (but I also didn't experience the issue with Qt5, so that doesn't really mean anything :) ).

closed 2025-02-09 by github-actions[bot]