FreeCAD/FreeCAD #10100: [Problem] Sketcher construction color + badground color are ill-suited for colorblinds.

PaddleStroke, 2023-08-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.22
```

### Subproject(s) affected?

None

### Problem description

I have the most common type of colorblindness and for my eye the default color of sketcher construction color is hard to see on the default background color. Here's a picture where I removed the red from the purple-ish color to give you a hint at how it looks blue on blue (I'm not sure if this sample will be very clear though):

[image]

So I suggest that we change either the background color, or the construction line color.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-07:
Would a colorblind mode (like in some video games) make sense or are there too many variants of colorblindness?

PaddleStroke, 2023-08-07:
I don't think this is necessary. Many program manage to avoid the necessity
of colorblind mode by choosing a palette of colors that are
colorblind-friendly.
Here there just this one that is annoying.

FEA-eng, 2023-08-07:
Right, it's rather poorly visible even for non-colorblind users. I could say the same about the dimensions' color. Theoretically, it contrasts with the blue background but in practice, it's not contrasting enough.

obelisk79, 2023-08-08:
When choosing colors, it would be wise to choose based on *contrast*.

Take the desired color palette and desaturate them against the background color, the resulting contrast of the image afterwards should give a good indication of appearance in instances of color-vision deficiencies.

chennes, 2023-08-08:
Maybe the first-run Wizard that @benj5378 is working on should have a "high-contrast" option... isn't that what Visual Studio does?

benj5378, 2023-08-08:
@chennes Good idea, but implementation wise, it would just be a theme/preference pack together with other themes.

chennes, 2023-08-08:
That makes sense to me, but I am also fine with changing the "classic" (ahem) background color. Just be prepared for loud complaints about it.

benj5378, 2023-08-08:
@chennes Friedrich Niche: classic is dead

PaddleStroke, 2023-08-08:
Colorblinds should not have to use the high-contrast option. The high-contrast migh be useful for monochromacy-colorblind, but those are like 10000 times less frequent than the regular 'red-colorblindness' that affects 8% of male people.

It is simple enough to accomodate the classic red-colorblind by choosing some colors rather than others. And I think @obelisk79 is right, choosing colors by contrast makes sure everyone can see them clearly.

Besides @FEA-eng seems to say that even for normal-eyed people the colors are not ideal. So maybe the color has to change. Either the background or the construction geo. Though the background seems to be illsuited for constraints as well for some, so maybe rather the background.

FEA-eng, 2023-08-08:
> Though the background seems to be illsuited for constraints as well for some, so maybe rather the background.

Right, it might be best (and easiest) to change the default background. Currently, dimension labels are poorly visible on it, especially when they are located within the darker regions:

[image]

obelisk79, 2023-08-08:
I am working on a full icon then using a palette of colors I've already tested for good contrast. I can make some proposals for the new light/dark themes which should work well without touching the 'classic' one. Would that be a viable compromise?

PaddleStroke, 2023-08-08:
I think the default should be fixed. Colorblinds shouldn't have to fiddle
colors or themes to have something acceptable.

obelisk79, 2023-08-08:
I can propose some color change suggestions.

obelisk79, 2023-08-08:
Below is an analysis of the current default gradient color against the entire palette of sketcher colors. It doesn't look right, but it isn't entirely apparent until you desaturate the image and see the glaring holes and poor contrast (shown in the green boxes)

If the goal is the *easiest* solution with minimal changes to the gradient. I suggest taking the bottom color and making it a variation of the top color with reduced saturation for the gray.
I found #454554 to be a reasonable color without changing anything else. The contrast overall is still poor in several cases but this should be considered for the new default "Light Theme" and "Dark Theme" rather than classic.
[image]

luzpaz, 2023-08-08:
Note: this is a duplicate of #6862

FEA-eng, 2023-08-09:
> Note: this is a duplicate of #6862

Only seemingly, here we came to the conclusion that instead of separate modes for colorblind people we need a change in the default colors.

chennes, 2023-08-09:
If I am understanding the discussion correctly, the proposal is to change the default background gradient to use a darker gray tone, as shown in this image from @obelisk79: 

[image]

This significantly improves accessibility without requiring a complete color scheme overhaul, according to the analysis above. I think this is a solid technical basis for the visual change, but would appreciate feedback from other Maintainers before we go ahead with this. @sliptonic, @WandererFan, @adrianinsaval, @0penBrain, @Roy-043 any thoughts?

FEA-eng, 2023-08-09:
@chennes To be honest, I'm not sure about it. I appreciate the technical analysis but the top part of that gradient seems to be pretty much the same as in the current background meaning that dimensions will still be poorly visible (my example above). Besides, we should consider aesthetic aspects too - how well the sketches and 3D models will look on this background. There was a discussion about changing the default background (not for colorblind people, just to make it look better) some time ago. I think that the conclusion was that we need a lighter shade. Maybe a more grayish background would work too. It might be best to prepare a few versions (considering visibility for colorblind and non-colorblind users as well as aesthetics) and ask people for their votes.

chennes, 2023-08-09:
A vote is fine for things that are purely aesthetic, but in this case there are technical considerations as well: at some point in the (distant!) past someone thought the current gradient was a good idea, after all 😄. I'd prefer in this case to find a technical solution to the problem first, then if it turns out we don't have an obvious choice and have reduced it to a pure "personal preference" decision, we reconsider whether a vote is appropriate. So if the above isn't good enough, let's keep working to find something that is, and see if the question answers itself.

obelisk79, 2023-08-10:
@FEA-eng holistically speaking, I believe some recent PR's are implementing a default Light and Dark themed preference pack which I think is where a more targeted review, assessment and overhaul of the entire color schemes should be conducted and push the aesthetic appearance into what many would call a more 'modern' look.

In this case, we are talking about the classic defaults of FreeCAD. My personal suggestion would be to either make the minimal change to address the visual impairment issue or just leave it as 'classic' and focus solely on the 'modern' ones where it will be less controversial to address.

PaddleStroke, 2023-08-10:
@obelisk79 The contrast is better, but to my eye this new proposal is actually worse. So it seems contrast is not all that is required.
[image]

For example the 'external geos' becomes virtually invisible to my eye. This is because I don't see red, so if you remove the red component from a pink it becomes gray, which in this case match pretty well the darker gray.

I think the problem relies on the fact that the computer computes the dark/white picture based on 3 color visions. So the contrast you are calculating by making the picture black and white is not the contrat color blinds are actually seing.

PaddleStroke, 2023-08-10:
Maybe for the colors we should take a hint from professional softwares such as the palete that visual studio is using. They most likely solved this colorblind issue to the best of their capacity, so there's no need to reinvent the wheel.
IMO we should have just a light theme and a dark theme by default and we should copy-cat the palete that is used in a widespread software such as VS or microsoft-office where they have put in a lot of efforts in solving that issue.

berniev, 2023-08-10:
No intention to be critical, but surely this business of color impact is not a new. Seems strange to be having a huge discussion of something that surely has been well researched etc ?? Surely there are papers, research, results, experts etc etc that mean we don't need to reinvent the whole wheel? Can't we defer to the experts?

0penBrain, 2023-08-10:
> @sliptonic, @WandererFan, @adrianinsaval, @0penBrain, @Roy-043 any thoughts?

Nothing special from my side. Roughly I don't care, maybe because I have no eye deficiency so not really a concern for me. Just avoid something too close from the default grey color of shapes. 😉

obelisk79, 2023-08-10:
For cases like paddle, I think the best approach is to leave the classic defaults as-is and develop a color conscious light/dark theme which is consistent with new preferences and considering a first run wizard being included in the next release where classic/light/dark will be selectable by new users etc.

berniev, 2023-08-10:
> we should copy-cat the palete 

IMO this is a naive approach in the sense that the pallette being copied has been designed for a specific environment/application/audience/fashion. You wouldn't blindly copy pasting code you don't understand. Having said that you gotta start somewhere.

obelisk79, 2023-08-10:
Not only that, many applications have palettes that are too limited in color selection to meet FreeCAD's needs. Sketcher alone is heavily dependent on color for various elements and constrained or not.

obelisk79, 2023-08-10:
Can anyone link a website that evaluates palettes for multiple types of vision deficiencies? I've been working on one and I'd like to ensure it is broadly accessible.

adrianinsaval, 2023-08-12:
Sketcher isn't the only thing to be taken into account when changing the background color, there's the measurement tools, Draft objects and annotations, the default part colors, the FEM widget, some stuff from Path.

From what paddle is saying I think it's clear we need to change the upper color and it seems the light grey from bellow doesn't work quite right either.

I suggest changing both default colors and forget about trying to keep the "classic" bs, if people like the current colours they can just keep their config and be done with it there's no reason for us to keep a bad color choice in code just because a subset of people like it.

MisterMakerNL, 2023-08-12:
I just made this issue because because it's nice to update the sketcher colors, but won't work if those colors are set in the object properties of the sketch and not the user settings:
https://github.com/FreeCAD/FreeCAD/issues/10180

MisterMakerNL, 2023-08-12:
We can btw always make a classic Freecad addon :)

obelisk79, 2023-08-14:
I have proposed what I believe are the minimal necessary changes to improve the classic colors for visual impairments and reduce eyestrain. I need feedback before fully endorsing these as good changes: https://forum.freecad.org/viewtopic.php?p=699524#p699524

obelisk79, 2023-08-15:
### Update:

I've been all over the spectrum trying to figure out how to get the contrast right, without completely changing the overall color scheme/feel of the default/classic style of FreeCAD. My prior suggestion work, but pose issues against flat solids where the light washes everything out to near-white.

However, this is an overall improvement over the current situation, and therefore I propose these changes move forward. In addition to that, I propose that in a separate issue/pr that FreeCAD extends the dark glow/blur used around dimension text to all sketcher lines/symbols. Here are some examples what this looks like without(this is mocked up in inkscape for visualization purposes):
[image]

Here, is with the proposed mild shadow effect (visibility improves greatly):
[image]

Here is with the shadow but no objects (looks normal):
[image]

obelisk79, 2023-08-15:
Expect a PR with color changes in a day or so.

drf5n, 2023-10-10:
> Below is an analysis of the current default gradient color against the entire palette of sketcher colors. It doesn't look right, but it isn't entirely apparent until you desaturate the image and see the glaring holes and poor contrast (shown in the green boxes)
> 
> If the goal is the _easiest_ solution with minimal changes to the 'classic' colors. I suggest taking the bottom gradient color and making it a variation of the top color with reduced saturation for the gray. I found #454554 to be a reasonable color without changing anything else. The contrast overall is still poor in several cases but this should be considered for the new default "Light Theme" and "Dark Theme" rather than classic. [image]

The default sketcher colors are hard-coded into files like:

https://github.com/FreeCAD/FreeCAD/blob/191aee568fcba9eb1149d9a66694a7b997715f4e/src/Gui/View3DSettings.cpp#L411-L413

and

https://github.com/FreeCAD/FreeCAD/blob/191aee568fcba9eb1149d9a66694a7b997715f4e/src/Mod/Sketcher/Gui/SketcherSettingsColors.ui#L44-L48

To be clear, would this be changing the hard coded defaults or the default Preference packs?
 https://github.com/FreeCAD/FreeCAD/tree/191aee568fcba9eb1149d9a66694a7b997715f4e/src/Gui/PreferencePacks

pierreporte, 2023-10-28:
I think this could be solved without the need to find new colors. Now that there are layers that allow for line styles, there should be less colors in general.

First, normal geometry should be bold, continuous lines like they are today, and construction geometry should be thinner, like engineering drawings. Second, construction geometry should be dashed so there is no need for two colors (if #7334 and its new axis line is implemented they can be distinguished without colors). Then there is a need for 3 colors only: fully defined, under-defined, and external geometry (currently it’s 5 for all this).

Points shouldn’t have a different color than lines but there should be two kind of points: normal geometry (a small circle like today) and construction geometry (a small cross) so no need for the red.

The dimensions could be the same color as fully defined geometry, which gives a visual clue. Reference dimensions could be the same color but with parentheses like in engineering drawings, which is makes their purpose more obvious. Two colors saved with this.

Constraints can, again, be the same color as fully defined geometry.

With this, the problem is way easier to fix. There is need for 3 colors only, plus a few extra for errors. There could be a different color for partially defined geometry (size defined but not position or the other). Solidworks uses only blue (for under-defined) and black (for fully defined). Constraints are black with a background but Catia is very readable already with just a slightly lighter green than fully-defined geometry. Note that, with what I describe, it is possible to show that lines are constrained but not their extremities, which is very useful.

Here is a color palette based on Catia, assuming a dark background:

| Element | Fully-defined | Underdefined | External | Over-constrained |
| ------ | :----: | :----: | :----: | :----: |
| Lines, circles, … | green | white | yellow | fuchsia |
| Points | green | white | yellow | fuchsia |
| Dimensions | green | N/A | N/A | fuchsia |
| Constraints | green | N/A | N/A | fuchsia |

pierreporte, 2023-11-13:
Here is an example of how it would look like. Note that the axis don’t have end points because we could imagine they are infinite and don’t need to be constrained that much.

[image]

tcurdt, 2023-11-13:
Changing the color of the dark background could help a lot.

But this would be my personal feedback:

- Fully-constrained and External is very hard to distinguish in Dark Mode.
- Incoherent in Dark Mode is nearly unusable
- Over-constrained in Dark Mode isn't great when dotted
- Fully-constrained in Light Mode is pretty bad

MisterMakerNL, 2023-11-13:
The background currently is gray or gray tinted because it conflicts the minimum amount of times with current (hardcoded) colors. And I'm not talking only about sketcher but all addons.

tcurdt, 2023-11-13:
> The background currently is gray or gray tinted because it conflicts the minimum amount of times with current (hardcoded) colors. And I'm not talking only about sketcher but all addons.

I was referring to https://github.com/FreeCAD/FreeCAD/issues/10100#issuecomment-1808874968

pierreporte, 2023-11-13:
@tcurdt This is exactly what Catia does. Indeed external geometry can be hard to distinguish. Incoherent geometry is rarely seen anyway and more a sign to hit Ctrl+Z anyway.

The color scheme should be tailored for each theme anyway. My propositions were more to show that we shouldn’t consider only colors but also line style to distinguish between the different kinds of geometry. As you can see, there is only five colors for 15 different line types so it’s way easier to find a readable set for each background color. If FreeCAD has two official themes (light and dark), there is only two sets to find. Any third-party need to reuse them or find a new one. A user willing to change has the responsibility of his or her choice.

Don’t hesitate to post examples of other CAD programs. For example I don’t know the color scheme of Creo or NX is (just that they have several line types).

obelisk79, 2023-11-13:
I disagree with the assertion that new colors aren't needed. If nothing else other than the aesthetic of using primary html colors being an issue related to the aesthetic-usability affect. Regardless, I think the proposal to reduce reliance on color and leverage the line variants mentioned above are a good idea.

I've already applied much of my color contrast and usability research in the OpenDark preference pack, general feedback regarding visibility has been positive and an OpenLight variant is being worked on.

PaddleStroke, 2023-11-13:
> Regardless, I think the proposal to reduce reliance on color and leverage the line variants mentioned above are a good idea.

+1 the number of colors we need is halved if we use dashed for construction. Which solves the palette issue and will make sketches look a little less like rainbows.

MisterMakerNL, 2023-11-14:
If you need any testers @PaddleStroke give us a shout!

abdullahtahiriyo, 2023-11-14:
I started to introduce visual layers as a more general concept of providing different groups of geometry different visible properties.

1) It could be dashing master sketches which have been carbon copied (or change their default colors), or providing thicker or thinner lines, so as to clearly tell apart segments belonging to the master and those built upon the master sketch. 

2) As a generalisation of that one, provide specific visualisation to a user layer (another concept WIP), so that groups of geometry imported via carbon copy are grouped together. Grouping together could perhaps enable improvements to carbon copy, such as adaptation to changes in the master sketch. Perhaps. It is an idea.

3) It could be changing construction to dashed, which has been requested many times.

4) It could be hiding and styling of internal geometry of complex shapes (e.g. axes of ellipses)

Building on top of that idea, is the also WIP a real proper model/view for geometry and constraints (and user layers and visual layers). That code is not even in master yet. This should allow to assign visualisation configuration to different layers easily.

Then you could also have automatic insertion into a given layer for specific geometry type or attribute (so construction assigned to dashed layer) (I would not like to have hard-wired options which diminish the flexibility).

However, this is not going to happen tomorrow (and it is basically only sketcher specific). I have a quite some dependent tasks to handle before that.

All that said, **I would strongly prefer to have a solution where colour codes types without dashing being involved**, as this gives further flexibility for other purposes.

I think colour blindness is important and we should do something about it, if only for inclusion and diversity (and a little bit of compassion). Because there is not a single type of color blindness, I tend to think that the best solution is to provide the possibility of configuring colour profiles, with some pre-installed colour profiles adapted for each type of colour blindness, and why not, provide the possibility to use your own choice of colours.

But all this post is written **without doing research**, and on this one I am quite with @berniev, we should look what is there first. 

While that is not done and for the sketcher, I posted in the referenced topic how to change colours for selection and preselection too from parameters. It is **not** ideal, but it is a possibility that exists today for a person having a problem today and wanting a solution today.

luzpaz, 2023-11-14:
> I think colour blindness is important and we should do something about it, if only for inclusion and diversity (and a little bit of compassion). **Because there is not a single type of color [...]

JFYI, #6862 references different types of color blindness.  
Here's Scribus's UI/UX approach: 
[image]

PaddleStroke, 2023-11-14:
@abdullahtahiriyo The correct way to implement layers (imo) is to have 2 kind of layers : coin layers and user layers.
Each user-layer is actually made of a set of 4 coin-layers : normal, construction, internal, external.
So user-layer layer1 has 4 coin-layers. And so in each user-layer you can have different types of geos. Then we could have a taskbox for layers. And enable customization of layers colors and so on.

I strongly prefer that construction and internal geos use dashed lines. For many reasons : 
- There is a consensus about it. Most people agree and it has been requested many times.
- Semantically dashed lines for construction make sense. All (?) softwares use dashed lines. Using something else goes against UX base principles.
- It reduce a lot the number of colors in sketches. Which makes things clearer and much better for colorblinds.

> Because there is not a single type of color blindness, I tend to think that the best solution is to provide the possibility of configuring colour profiles, with some pre-installed colour profiles [...]

I'm colorblind and I'm against specific color scheme. The software should be usable without further configuration. As right now I can already configure the colors. For me the solution is to reduce the color palette by using dashed lines + use colors that arecolorblind friendly. I think overall it just need to tweak a few colors. But if we want to support all types of colorblindness we could use special palettes : 
Here're some examples of colorblind palette : https://davidmathlogic.com/colorblind/#%23D81B60-%231E88E5-%23FFC107-%23004D40

abdullahtahiriyo, 2023-11-14:
@PaddleStroke 

I want to avoid forcing every user to use dashed lines for that specific purpose, while allowing users who want to use them for that specific purpose to use them.

In other words, if a colorblind person, or a regular user, wants to use dashed lines to reduce colors, fine with me, but this should not come at the expense of removing flexibility from other users, who may prefer the rainbow version and use dashed for other purposes (in my mind master sketches, as it is my workflow).

There is a finite number of combinations of workable line visualisations [hidden, dashed, thickness, color]. I have other possible uses for them (such as marking different user layers, which are groups of geometries).

This is the main reason why I am advocating to have a palette of colors. Perhaps there are colorblind people that would want to use dash for these other purposes (even from time to time), and who still would like to have acceptable colors when choosing to do so.

obelisk79, 2023-11-15:
@abdullahtahiriyo I can appreciate your intention with retaining legacy customization, but let me put into perspective what is happening with sketcher and FreeCAD, with regards to colors.

Sketcher WB has 21 customizable colors. Across a default installation of FreeCAD there are roughly 83 defined colors, not including hidden parameters. I think we can make several inferences from this knowledge.

1. 83 independent color definitions is unmanageable (sketcher accounts for 25% of them)
2. There is a possibility for "too much" customization
3. There can be made a strong case for implementing alternative visual clues instead of color.
4. Color-blindness, accessibility are nearly impossible to effectively manage with 21 colors defined in sketcher.

Some of these items are outside the scope of this github issue. However I believe there is both room for simplifying color usage and leveraging line patterns. This, in my mind, is a bigger problem than customizing the appearance of geometry by user-selectable layers. 
The Hick-Hyman Law of UX would imply that we should minimize customzations in favor of more sensible default settings and controls.

https://github.com/FreeCAD/DevelopersHandbook/blob/main/designguide/laws-of-ux.md#hick-hyman-law

PaddleStroke, 2023-11-15:
@abdullahtahiriyo I am not saying we should remove support for colors of construction lines. In the settings we can have the possibility to customize color, pattern and thickness of each type of geometries. In the same way that I implemented for the grid lines.

My argument is for the default style/color of lines.

abdullahtahiriyo, 2023-12-09:
@obelisk79

I am all in for sensible defaults. 

However, one should realise that if 21 colors in the sketcher are different, and this difference is purposeful from a functional point of view, "reducing the number of colors" means either reducing the functionality or taking coding of the different functionality by using something that is not color (if you remove one bit from the former and keep the functionality, you need to take one bit from elsewhere). This necessarily takes away that other "something" from using it for another purpose.

If coding 21 colors into a scheme is way too much as you claim (Paddle certainly knows better than me), we should by all means help there by taking advantage of those other visual hints. However, what I mean is that, we should not take the functionality away from other users.

Then the argument is whether we should make colorblind settings default and allow users to deviate, or whether we should make reasonable defaults for the target 100% population and allow colorblind people to deviate (e.g. by activating a profile). 

There will be settings were nothing is gained or lost. Most users will not care much one colour or another for a given function. Then, I think it is reasonable to set the one that is good for colorblind (considering that there are several types of colorblind), if one is better than the other. Then the default is good for most. Who does not want can just make an own customisation.

There will be settings where regular users lose flexibility if we take colorblind defaults. In that case I do not think colorblind defaults are appropriate. Yes, every user can customise, but we end up with 80% of the normal population having each his/her own customisation which is clearly suboptimal.

tcurdt, 2023-12-09:
I think this issue could maybe seen as a good devils advocate.

1. If 21 colors are really needed, maybe there is problem beyond color blindness? Maybe that's just too much state differentiation - but that makes it a bigger issue. I personally found the idea of dashes and shadows a great quick win.
2. Given that that color impairment is a spectrum - maybe not all of the unimpaired people can easily distinguish all those 21 colors either? It should be expected that a colorblind profile will also make it easier for the non-impaired.

obelisk79, 2023-12-09:
Well, we could challenge ourselves on a few of the easier ideas. Do points/vertex need a separate color? Is there a correlation with other similar software? Color to indicate constrained elements or not is good, but if each element changes color when defined with constraints, do we truly need a separate color to indicate 'fully constrained' geometry/construction? These are some areas where a simplification may actually be easier without a loss of meaning. I have provided many examples in previous posts to this discussion with a simple analysis of just the color contrast issues which stem from sketchers many colors used. There is even more here: https://forum.freecad.org/viewtopic.php?t=80355

I have proposed new default colors which address the majority of color blindness cases without reduction in color use. However, there is still room for proper visual distinction without so much color being used which can become quite visually distracting in complicated sketches. Theme designers additionally have to struggle with this conundrum of matching, effective colors. The emphasis should be on geometry, construction/reference elements should be muted to allow the mind to focus on the essence of the work. I suppose one of us could conduct a meta-analysis comparing freecad to other industry software so that we can assess if what exists in FreeCAD is more effective than what is seen among the majority.

These large companies spend large amounts of money on UI/UX professionals, analytics, and use working groups to determine what is necessary and effective while interacting with their software. This massive amount of work is evidenced in that software. FreeCAD lacks those resources, however we can learn from studying the results of their efforts.

obelisk79, 2023-12-09:
@abdullahtahiriyo 
Well, we could challenge ourselves on a few of the easier ideas. Do points/vertex need a separate color? Is there a correlation with other similar software? Color to indicate constrained elements or not is good, but if each element changes color when defined with constraints, do we truly need a separate color to indicate 'fully constrained' geometry/construction? These are some areas where a simplification may actually be easier without a loss of meaning. I have provided many examples in previous posts to this discussion with a simple analysis of just the color contrast issues which stem from sketchers many colors used. There is even more here: https://forum.freecad.org/viewtopic.php?t=80355

I have proposed new default colors which address the majority of color blindness cases without reduction in color use. However, there is still room for proper visual distinction without so much color being used which can become quite visually distracting in complicated sketches. Theme designers additionally have to struggle with this conundrum of matching, effective colors. The emphasis should be on geometry, construction/reference elements should be muted to allow the mind to focus on the essence of the work. I suppose one of us could conduct a meta-analysis comparing freecad to other industry software so that we can assess if what exists in FreeCAD is more effective than what is seen among the majority.

These large companies spend large amounts of money on UI/UX professionals, analytics, and use working groups to determine what is necessary and effective while interacting with their software. This massive amount of work is evidenced in that software. FreeCAD lacks those resources, however we can learn from studying the results of their efforts.

pierreporte, 2023-12-09:
> No other software I have ever used maintains such a thing, they merely design with this in mind if needed.

They don’t because they don’t need it in the first place. They don’t have 21 colors to differentiate. Solidworks has basically 4: black (constrained geometry and constraints themselves), blue (unconstrained geometry), green (background of constraints), red (for errors). No specific color for points and constraints. They use thickness and patterns.

It’s hard enough to precisely distinguish between 21 colors without any disability in a perfect environment but it is necessary to take into consideration that there now is screen dimmers (the display turns reddish in the evening) and that screens can display colors differently (very few people calibrates their monitors).

I want to add that using different line weight and line styles to differentiate the function of lines is the normal way of drawing in engineering and geometry in general. It will feel natural.

MisterMakerNL, 2023-12-09:
> These large companies spend large amounts of money on UI/UX professionals, analytics, and use working groups to determine what is necessary and effective while interacting with their software. This [...]

That is why we kinda "learn"from them instead of trying to do it different and better! So I vote for as using colors only as a last resort and try to get these layers and line types up and running. Also we can always color layers instead of lines just like in autocad.

abdullahtahiriyo, 2023-12-10:
I will not repeat myself. I will just mention that:

1) If software A codes less states, perhaps it also offers less amount of information. If it does perhaps it is worse and not better (or at least for users which do want that extra information).
2) If it does not code less amount of information, then it needs take "bits" from to other information (thickness, patterns). Then this other kind of marking stops being available for marking other features.
3) If we do not have other markings available, then the usability of future layers functionality will decrease considerably, because precisely, thickness and pattern is one way to differentiate different layers seamlessly. This is because dashes and thicknesses are orthogonal to colour in the current state, i.e.  the same color code can be used for the different layers consistently, but the user can clearly differentiate the layer the geometry belongs to by the dash pattern or thickness.

Now, when layers is available, you could have a layer for construction geometry (if that is what ticks for a given user). If one wants to have that layer with a discontinuous pattern it is just fine. The person will have a different color (construction color) and a different pattern. (User A use case)

At the same time, if one is using carbon copy, this user can just set the carbon copied geometry (from other sketch) into a user layer, and assign a discontinuous pattern to it. This way, it will be very easy to differentiate the carbon copied geometry from the own sketch geometry. (User B use case)

By not changing the colors, user A and user B can perfectly understand all the current signaling just from the color. This means that user A understands all the color-coded common information from the sketch of user B and vice versa. User A can have a different pattern to "offload" color-only differentiation, whereas User B can add more signaling relevant to him. (Of course, that is if none of the users changes colors)

Now, some users appear to suggest to remove the colors and user the patterns instead. Then there is no "common color information" signaling that is orthogonal to pattern/thickness between user A and user B, because the moment user B (even when allowed), tries to use dashes differently, then user A does not understand what is construction and what is regular geometry, because it is not using the same color (because of the "mandated" reduction of colors). In fact, even user B will be unable to differentiate them.

In fact, having the choice to have 21 colours does not mean having necessarily 21 different colours if one does not wish to. If somebody does not need vertices with different colour, just configure the same colour as edges. If somebody does not need to differentiate fully constrained sketch, just change the color to the normal one. Today, one can set edges and vertices the same color, constrained or not the same, external or not the same. constraints the same as edges. If you want just one colour, you can have it too, just go on and make everything white. If there is enough people who want to have one such profile with reduced colours, we can just make  a combobox of profiles. If there are such fans of the 4 colour only CADXXX, a profile can be created that configures it like that. **All I ask is please not make it the default**. The current colorful version can be called "FreeCAD" and be the default. Full freedom, but respect for the others.

Even, when layers is implemented, then you can have your rule to add construction to a layer with dashes, and then if only "white" colour is configured for everything, at least construction will be dashed... full freedom again.

This is what I mean when I say: Yes, by all means, let people have what they want (or need), **but do not take away opportunities from the rest**. 

To put one analogy: 
We have people in wheelchairs. We improve accessibility by building ramps (which is a very good thing). Still many newly built buildings (if not all) do build regular stairs in addition to necessary ramps. In fact, we still build internal staircases in (high) buildings with all accessibility certifications (usually close to the elevators).  We have no laws forbidding stairs. We still allow people to use stairs (often even when ramps are available). The proposal equivalent is some kind of ban on staircases, forcing the building of ramps and forcing everybody to use them wheelchair or no wheelchair.

Another analogy:
I am relatively short. When I play basketball I do not ask others to play on their knees to compensate. At the same time I am thankful others do not ask me to play on high-heel shoes...

pierreporte, 2023-12-10:
@abdullahtahiriyo At the end of the day we have the same opinion: let users decide what they want to do. Except that I am biased in favor of having less colors and more line styles. The preferences should allow to define the color, thickness and pattern of every line type, dimensions and points in every configuration.

> If there is enough people who want to have one such profile with reduced colours, we can just make a combobox of profiles. If there are such fans of the 4 colour only CADXXX, a profile can be [...]

This is a good proposition. Say we have two profiles:

* classic FreeCAD;
* industry standard.

The industry standard profile could be based on Catia, which is the easiest because of the current theme (dark background). Then you and other users will test it and determine what is best for their use of FreeCAD. It would be a good start to chose what the default would be. And, as you said, it wouldn’t forbid users to switch back to colors only.

An issue could be created right now as this one is too broad (and should be turned into a discussion).

Nobody is asking to remove colors altogether. The 3 possible variations would be used at the same time.

I have a hard time understanding your use case of layers. As a professional CAD user, I never felt the need to have layers in a sketch. When I have complex shapes overlapping, which is very rare, I often create several sketches on the same plane and I can reference other sketches in the current sketch (without using an equivalent to `Sketcher_External` so the view isn’t polluted). Can you describe what you are trying to do?

obelisk79, 2023-12-10:
I find the description of layer usage to be confusing and more theoretical than it is practical. I think there is some value in layers, but I believe we have vastly differing opinions of that actually means.

I view the potential of layers as a way to organize the elements of a sketch internally. I envision as follows:
1. Each element type is drawn on its own layer (point/vertex, geometry line/arc, reference/construction line/arc, symbology)
2. Layers are then used to create draw elements with different style (patterns)
3. Layers are given selection priority logic (points > lines/arc > symbols)

This could be configurable in preferences (not on the toolbar). The user should not be exposed to layer management while working. This would be such a highly niche thing which may have some interesting uses but would not be broadly useful it isn't worth exposing to the user. My vision outlined above would be broadly useful to the vast majority of users, veterans, newbies and everyone in-between.

sliptonic, 2023-12-10:
> let users decide what they want to do.

This is only true to a point and then it rapidly becomes a negative.  For some light reading, google for 'tyranny of choice'

The summary is that too much choice makes users miserable.  You've heaped a burden on them to first understand what all the choices imply, and then to select from them.  It's orders of magnitude worse when the available options interact in unpredictable ways.  Even if they are willing to invest the time and energy to configure the application, there's a nagging fear that they've missed something that will make them happy.  Opportunity cost sucks life.

Choice is a good thing for user satisfaction but it must be used carefully.  I'm most bothered when developers are facing difficult choices about UX and avoid making those choices by just passing it off to the user as though enabling one more preference setting is always a good thing.

It's much harder to understand what makes a good user experience and build it.  It's easy and lazy to just dump the choices on the user.

[24 more replies]