FreeCAD/FreeCAD #10101: [Problem] Sketcher : Changing the layer of a geometry to layer 1 is undone if we toggle the visibility of said geometry

PaddleStroke, 2023-08-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Problem description

Change a geometry layer from 0 to 1. The geometry becomes dashed.
Then in element taskbox toggle the visibility of the geometry. Then the geometry is back on layer 0 (undashed).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-08-07:
I can confirm this.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

PaddleStroke, 2024-02-14:
Fixed with https://github.com/FreeCAD/FreeCAD/pull/11996

closed 2024-02-14 by PaddleStroke