FreeCAD/FreeCAD #10107: [Problem] Floating point issues are hard to troubleshoot

Jookia, 2023-08-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: makepkg
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.5
  * Manipulator 1.5.0
  * fasteners 0.4.56
  * Help 1.0.3
  * Render 2023.2.3
  * POV-Ray-Rendering
```

### Subproject(s) affected?

None

### Problem description

FreeCAD uses floating point operations internally, and rounds it at the GUI level. It doesn't explain this to the user which can lead to extremely strange situations.

Here is an example: [test.zip](https://github.com/FreeCAD/FreeCAD/files/12285676/test.zip)

In it contains a padded triangle with holes along two of the angled sides.

- The middle plane is offset by ```sqrt(26450) / 2```
- The initial hole is offset by 36.25mm
- A multitransform creates a linear pattern of length 42.50mm
- The multitransform then mirrors the pattern around the middle plane

This should lead to a triangle with holes on each side but in this specific case the transform fails due to the mirror step. Changing any of these three magic numbers (including the sqrt) fixes the issue.

In this case these values were found in the wild when making a more complicated part using a spreadsheet and many more expressions. This was difficult because:

- All values shown were rounded by the GUI
- Rounding the final value didn't help because the error is introduced elsewhere
- The final broken value is likely inside FreeCAD's core logic, not visible to the user

It would be nice if FreeCAD warned on values that aren't rounded to a certain degree or automatically rounded them for us to avoid this situation.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-08-28:
> It would be nice if FreeCAD warned on values that aren't rounded to a certain degree or automatically rounded them for us to avoid this situation.

I do not see what we could do here at all. How is the program supposed to make decisions that are on a case by case basis? I doubt that rounding is really the problem here, this seems like an upstream bug in occt with coincident cylinders.

Jookia, 2023-08-29:
FreeCAD already visually rounds the numbers, making this extremely hard to troubleshoot. Maybe disable that for now?

adrianinsaval, 2023-08-29:
The number of decimals shown is configurable. Disabling this by default is not an option IMO, we can't be showing 16+ digits to the user all the time.

A case could be made about increasing the default number of decimals. But again I don't think the number of decimals is really the problem here, most likely it's the coincident cylinders.

Jookia, 2023-08-29:
How would the coincident cylinders be the problem?

adrianinsaval, 2023-09-12:
the geometric kernel (opencascade AKA occt) often struggles with boolean operations involving coincident geometries, often referred to as the coplanar issue although as can be seen here it also affects coincident curved surfaces/edges

adrianinsaval, 2023-09-13:
[image]

in your model the original holes and the mirrored holes intersect and are coincident here, I highly suspect the kernel bugs out in this situation.

I get the model to actually work by displacing the mirror plane by 0.0001 so that it is not coincident anymore.

Jookia, 2023-09-13:
How do I move this bug forward to OCCT?

Jookia, 2023-09-13:
I did some more research and this seems like a fundamental issue with CAD software and mathematics?

adrianinsaval, 2023-09-14:
yeah I don't think this issue is actionable so I'm leaning towards closing it.

Jookia, 2023-09-15:
Is there a way to warn when things are coincident?

adrianinsaval, 2023-10-13:
what kind of warning are you looking for? I can't think of much more we could do about this here. Since this doesn't seem actionable I'm leaning towards closing this issue.

Jookia, 2023-10-13:
Is there a way to detect boolean conflicts like this? Does check geometry handle it?

adrianinsaval, 2023-10-16:
check geometry can help you identify this but it can just as well give you false positives, if you need more information for your own modelling practices open a thread in the [help forum](https://forum.freecad.org/viewforum.php?f=3) or search the web.

I'm closing this issue as there's no reasonable action for us to take here.

closed 2023-10-16 by adrianinsaval