FreeCAD/FreeCAD #10108: [Problem] Cannot double click to edit constraint

andyylin, 2023-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git)
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Launcher
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

I recently switched my main machine from Mac to Windows. I found that on Windows, I am unable to double click on an existing constraint to edit it. I have to triple or quadruple click rapidly for the window to pop up. Seems like this "bug" (?) has been around for a while, and I don't see it in the Github open issues.

### Anything else?

There was a forum thread in 2022 that discussed this issue: https://forum.freecad.org/viewtopic.php?style=1&t=71137

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

G0MJW, 2023-08-12:
I have the same issue. Sketcher is completely unusable as it seems impossible to edit a constraint. This was not an issue on 0.20. I have wasted an hour trying to get this to work.

petrrpancon, 2024-01-16:
version 0.21.1 still the same problem on intel i7 ~4GHz notebook not fast enough to register double click. It gets better when i slow the system double click speed. Obviously I wont do that every time I switch to different app window :-). I have multiple apps based on qt and none of them has this problem. Looks like the problem is caused by measuring time do de-bounce mouse button in UI thread which is terrible idea. I am surprised that QT would not provide the mouse click event following system setting.

FlachyJoe, 2024-03-09:
@maxwxyz I can't test with Windows so I can't go further about this bug.

maxwxyz, 2024-03-09:
I cannot reproduce this. @petrrpancon @G0MJW @andyylin could you try this in the lates weekly https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Please also provide a test file where this error occurs.

G0MJW, 2024-03-09:
Can you remove me from this update list - this was something from back in August.

andyylin, 2024-03-11:
> I cannot reproduce this. @petrrpancon @G0MJW @andyylin could you try this in the lates weekly https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds Please also provide a test file where this error occurs.

Still exists in this version. Nothing happens when I double-click the radius constraint value, but the edit dialog window pops up if I triple click.

[test.zip](https://github.com/FreeCAD/FreeCAD/files/14552458/test.zip)

andyylin, 2024-03-11:
> Can you remove me from this update list - this was something from back in August.

I believe you need to unsubscribe from this thread.

[image]

maxwxyz, 2024-03-11:
> Still exists in this version. Nothing happens when I double-click the radius constraint value, but the edit dialog window pops up if I triple click.
> 
> [test.zip](https://github.com/FreeCAD/FreeCAD/files/14552458/test.zip)

I cannot reproduce this with
<details><summary>Windows 11 0.22 dev</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.2
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.12
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.2

```

</p>
</details>

petrrpancon, 2024-03-18:
Hi, this problem is likely a combination of more factors. I do not have the problem on my box PC with nvidia, but only on notebook. It is dual GPU (intel integrated + amd). It could be also affected by another background application. It seem I have to click faster in to edit the constraint then anywhere else to be accepted as double click in even in the dev version 22. Unfortunately I never found the common factor while was broken. I have a new notebook now with only integrated gpu. I have not seen it happen there so far even with 21.1. So basically you have to be "lucky" to have a system where it happens to debug it :-/.

maxwxyz, 2024-04-13:
Maybe related:
- https://github.com/FreeCAD/FreeCAD/issues/13439

luzpaz, 2024-08-14:
@chennes can you reproduce on windows ?

petrrpancon, 2024-08-14:
Yes, on my new computer ryzen 7 4700U I can edit the constraint length, but I still have to double click much faster than in any other app. It is very clear when comparing to windows mouse setting panel that has the double click test area. Possible workaround is triple click. That seems to work even with system click speed. I have tried latest weakly build and release 21.2 there is no difference related to this.
I have the double click set to faster than default. I think there is some additional delay making it bigger problem as the click interval is set to smaller values. With default setting the time is so long that the delay is causing only small error.
I suggest you to set the double click in windows fastest possible you can reliably use and then try to click edit the constrain in the same passe and it is immediately obvious it does work unless you click much faster

petrrpancon, 2025-02-17:
Is this a joke? :-o. Closing unresolved issues to "keep our backlog tidy" is absurd and great example why is this SW so unstable. 
Either fix it or accept it as unfixable due to bad SW design or what ever is the real reason.

maxwxyz, 2025-02-17:
@petrrpancon feel free to create a PR to fix it, I'd be happy if all 2.600 issues are resolved. This particular issue is not reproducible by many and won't affect the majority of users, so there is not more we can do until to wait that someone will pick it up and fix it in their spare time.

petrrpancon, 2025-02-17:
@maxwxyz  I can understand you wont spend time on it it it affect small portions of users. I was referring to "keep our backlog tidy". That is exact opposite of thy the tracking systems exist! So i stand by the logic that it is absurd reason ;-)
Feel free to close as unresolved, at least others will be able to see it as known problem for some.

maxwxyz, 2025-08-25:
@petrrpancon can you check on the dev version: https://github.com/FreeCAD/FreeCAD/releases

chennes, 2026-09-11:
This bug is a zombie, only still alive from manual pings that get no response. I'm closing it.

closed 2026-09-11 by chennes