FreeCAD/FreeCAD #10109: Pocket Path does not generate toolpath for pockets in non-uniformly scaled models

mwiktowy, 2023-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 41b058e)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * lattice2 1.0.0
```

### Subproject(s) affected?

Path

### Problem description

Steps to reproduce:
I made a model from imported SVG geometry: One sketch formed the profile and one was the offset pocket in it.

The model was slightly too small to maximize the use of the width of the board that I wanted to mill. I scaled the X and Y of the model (either using scale properties of the Draft Clone or in the Job model) and tweaked the X/Y scaling factor to fit the width of the board but keeping the Z scale at 1.0.
Pocket operation could not find the selected bottom face of pocket as a reference. It gave the following error:
22:38:36  PocketShape.ERROR: Pocket does not support shape Model-Cut002.Face74
22:38:36  Geom.INFO: Zero working area to process. Check your selection and settings.
22:38:36  <Path.Area> Area.cpp(1429): hit bottom -0,0,1e-06
22:38:36  <Path.Area> Area.cpp(1429): hit bottom 0,0,1e-07

3D Pocket generated a pocket toolpath but did not recognize the pocket floor and generated a path that went clear through the model and specifying a Final Depth was not possible for that operation.

Profile Operation could follow the inside of the pocket (referencing the bottom edge of the pocket) and the outside of the profile (referencing the outer edge of the model) of the scaled copy.

If I scale X/Y/Z uniformly, the Pocket Path will generate a tool path properly.

Now that I understand (after a lot of confusion) what caused a toolpath not to be generated, there are work-arounds by manually setting a Final Depth that compensates for the excess scaling in the z direction but it would be nice if this was not needed.

### Anything else?

Thanks for the great work on this Path workbench. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwiktowy, 2023-08-08:
[Example.zip](https://github.com/FreeCAD/FreeCAD/files/12292568/Example.zip)

Made simplest case example file to demonstrate the issue. Change scale of "Model-Body" in Job non-uniformly and Pocket Operation will fail but Profile Operation succeeds.

JohnOCFII, 2025-01-16:
This is still an ongoing issue.

sliptonic, 2025-01-17:
I'm wondering if this is a lingering toponaming problem.  The example file is a good one.  I can duplicate the issue but for me it breaks both the pocketing and profile operations.  Error message seems to indicate a failure outside CAM workbench.
I've pinged @chennes

maxwxyz, 2025-07-27:
@drwho495 is this toponaming related?

drwho495, 2025-07-27:
> @drwho495 is this toponaming related?

Tarman3 made a pr to fix it, and just by looking at the description, no I don't think so. The error messages don't seem to indicated a TNP either