FreeCAD/FreeCAD #10113: [Problem] Path workbench does not work in FreeCAD 0.21.0 AppImage

jes, 2023-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (Regolith:GNOME-Flashback:GNOME/regolith)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.4.2
  * Render 2022.4.0
[/code]
```

### Subproject(s) affected?

Path

### Problem description

When I try to open the Path workbench I get an error box popup, with the following in the console:

[code]
Invalid version: 'camotics: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory'
Traceback (most recent call last):
  File "<string>", line 159, in Initialize
  File "/tmp/.mount_FreeCAlXdh3y/usr/lib/python3.10/site-packages/packaging/version.py", line 52, in parse
    return Version(version)
  File "/tmp/.mount_FreeCAlXdh3y/usr/lib/python3.10/site-packages/packaging/version.py", line 198, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
[/code]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-08:
@adrianinsaval @sliptonic both your input would be useful.

adrianinsaval, 2023-08-08:
where there new dependencies added to path during the 0.21 dev cycle?

sliptonic, 2023-08-08:
Nope.  But will use camotics if it is found.  https://github.com/FreeCAD/FreeCAD/pull/6637

adrianinsaval, 2023-08-08:
what is camotics? should it be shipped with the appimage? @jes do you have it locally installed in your system?

jes, 2023-08-08:
Ah! So the issue is I have camotics installed and it is broken. I can confirm that if I run "camotics" at the terminal I get the same error. I've uninstalled camotics and all is fine now.

Curious why this is a problem with 0.21 and not 0.20.

adrianinsaval, 2023-08-08:
presumably 0.20 does not try to use camotics

sliptonic, 2023-08-08:
No.  It's a manual install.  At least I don't think camotics has rolled an official version later than 1.20.
If it isn't already installed and sufficiently new, it should fail silently

https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Path/Path/GuiInit.py#L86-L99

adrianinsaval, 2023-08-08:
It fails silently with these exceptions:
https://github.com/FreeCAD/FreeCAD/blob/68d22d864b801231c83c0ef961073ac5f8764aa4/src/Mod/Path/Path/GuiInit.py#L98

but here it's raising `InvalidVersion` instead

jes, 2023-08-08:
It's because the code unconditionally passes the stderr of camotics into `packaging.version.parse()`, which fails with `InvalidVersion` if the version number is invalid (for example if camotics produces an error message on stderr instead of a version number).

sliptonic, 2023-08-08:
We can add the InvalidVersion exception to the list.

sliptonic, 2023-08-08:
[image] [Camotics Invalid version](https://trello.com/c/Zt0ElHyG/2-camotics-invalid-version)

closed 2023-08-19 by sliptonic

wolfgangr, 2024-05-02:
may I present a variant of the message:

```Invalid version: 'camotics: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/.mount_FreeCAqD1n4J/usr/lib/libnsl.so.1)'```

details see
https://forum.freecad.org/viewtopic.php?t=86609&start=10

living on some prehistoric debian (never touch a runing system ....)
... but at least, I have a working camotics running from the command line:

```
$ cat /etc/debian_version 
10.13
$ camotics --version
1.2
```

afaik, camotics does similiar bundling of runtime environment, as AppImage does.
So by halfway educated guess, may be they get it right to bed themselves, while FreeCAD AppImage builder still may face some challenge to prepare a ecosystem that suits camotics bare library???

commit 6ab40695 (2023-08-19): Merge pull request #10140 from Ondsel-Development/camoticsbug

fixes #10113
commit cc603e27 (2023-08-19): fixes #10113

packaging.version.parse() will raise InvalidVersion in some cases but the error isn't caught.
commit da2bc3ba (2023-08-30): fixes #10113

packaging.version.parse() will raise InvalidVersion in some cases but the error isn't caught.