FreeCAD/FreeCAD #10122: error after  freecad start

KAPO62, 2023-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
DirectWrite: CreateFontFaceFromHDC() failed (Gibt einen Fehler in einer Eingabedatei an, zum Beispiel eine Schriftartdatei.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=120
```

### Subproject(s) affected?

None

### Problem description

DirectWrite: CreateFontFaceFromHDC() failed (Gibt einen Fehler in einer Eingabedatei an, zum Beispiel eine Schriftartdatei.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=120

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-08:
To assist in others finding this in future, could you please amend the title to something that includes _8514oem_ as this is the critical word in finding the solution which is https://forum.freecad.org/viewtopic.php?style=1&p=654799#p654799
The types of fonts allowed for the Macro Editor, Report View and Python Console have been slimmed down from previous versions.

G0MJW, 2023-08-12:
It is happening to me with a clean install of 0.21 in Windows 11

21:21:27  DirectWrite: CreateFontFaceFromHDC() failed (Indicates an error in an input file such as a font file.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=144

I think this might have happened because I had to make the sketcher font larger as it is far too tiny to read with a high DPI monitor.

I see several posts but no actual solution, apart from a reference to an editor menu in preferences which does not appear.

OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.5.11
  * kicadStepUpMod 10.14.7
  * pcb

luzpaz, 2023-08-13:
What about the font is problematic ?

G0MJW, 2023-08-13:
I don't know - apart from that is doesn't appear to exist.

G0MJW, 2023-08-13:
Ah - there is a new menu in a different place - unexpectedly under python ! Hasn't yet fixed the problem but might be a way towards it.

[image]

Syres916, 2023-08-13:
@G0MJW I apologise, when I wrote the above post I completely forgot that during the 0.21 development cycle the user interface for preferences had significant changes made including this combobox location. It can now be found under _Python>Editor>Display Items>Family_. Here is a screenshot from my 0.21.0 build. On Windows, something like _Courier New_ should be acceptable and stop the warning. See screenshot:

[image]

G0MJW, 2023-08-13:
Thanks – I found this myself at the same time…

 

 <https://user-images.githubusercontent.com/46537884/260292904-a23f9be1-8e17-4bf4-8cda-f5b42b517156.png>

luzpaz, 2023-08-25:
Can anyone reproduce? 

Is the font proprietary? If not, can you upload it to this ticket ? (open github, zip the font file and drag/drop in to ticket)

luzpaz, 2023-09-04:
soft bump

Syres916, 2023-09-05:
@luzpaz I've made an attempt, can't think of a better way to handle it especially as I can't replicate it. I had to use an emoji font as my exclude test.

closed 2023-09-25 by yorikvanhavre