FreeCAD/FreeCAD #10123: Recent clang-format change is harder to read

berniev, 2023-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
na
```

### Subproject(s) affected?

None

### Problem description

#9994 Introduced a welcome change to break up long parameter lists onto separate lines, but introduces positional  inconsistency between long and short parameter lists. Every little inconsistency makes things just that little bit more difficult.

```
    int autoConstraint(
        double precision = Precision::Confusion() * 1000,
        double angleprecision = M_PI / 20,
        bool includeconstruction = true
    );

    int autoConstraint2(double precision = Precision::Confusion() * 1000);
```

Setting `AlignAfterOpenBracket: Align` would produce
```
    int autoConstraint(double precision = Precision::Confusion() * 1000,
                       double angleprecision = M_PI / 20,
                       bool includeconstruction = true);

    int autoConstraint2(double precision = Precision::Confusion() * 1000);
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-08-09:
Especially for the implementation of a function I find this style difficult because it makes it hard to quickly distinguish the function arguments from the function body:
```
void someLongFunction(
    std::vector<std::vector<int>> argument1,
    std::vector<std::vector<int>> argument2
)
{
    auto test1 = argument1;
    auto test2 = argument2;
}
```

I also vote to restore `AlignAfterOpenBracket: Align`

berniev, 2023-08-11:
PR #101167

closed 2023-08-11 by berniev