FreeCAD/FreeCAD #10126: [Problem] Default installation installs gtest/gmock header files

StefanBruens, 2023-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
n/a (build issue)
```

### Subproject(s) affected?

None

### Problem description

FreeCAD builds its own copy of gtest and gmock, and then installs it.

This causes several problems:
- automatic dependency generators may pick up the private libraries and auxiliary files (CMake config, pkgconfig, ...)
- FreeCAD binary packages ship completely unnecessary files (headers, auxiliary files)
- gtest is rebuild on each FreeCAD rebuild

Preferably, gtest/gmock is added as an CMake ExternalProject, and defaults to a system provided gtest/gmock.

### Anything else?

https://github.com/FreeCAD/FreeCAD/blob/292196a606be906202957b8693fd8d6d8db39a20/tests/lib/CMakeLists.txt#L28

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

berniev, 2023-08-09:
Great to see someone new here with (hopefully) knowledge and experience with cmake. Real cmake skills, especially modern cmake, are a bit thin on the ground.

StefanBruens, 2025-01-17:
This is probably solved mostly with
https://github.com/FreeCAD/FreeCAD/commit/11fa271eb53c99e56bbaa536fbd50ff308ae2d14

For packagers, it is possible to delete the `gtest` directory (shipped with the tarball) and then use the system gtest.

Preferably, FreeCAD had a global CMake setting to prefer system deps (or even force them).

closed 2025-06-17 by github-actions[bot]