FreeCAD/FreeCAD #10130: Segmentation fault when clicking Ctrl-Z

rezahousseini, 2023-08-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Guix System (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.Unknown
Build type: Release
Python 3.10.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.56
  * Assembly4 0.12.7
```

### Subproject(s) affected?

None

### Problem description

I have a TechDraw from an imported CSG file where I deleted some cut out views and wanted to go back to them by clicking Ctrl-Z. FreeCAD crashed with the following backtrace:

```
Unhandled std::exception caught in GUIApplication::notify.
The error message is: basic_string::_M_construct null not valid
Program received signal SIGSEGV, Segmentation fault.
#0  /gnu/store/gsjczqir1wbz8p770zndrpw4rnppmxi3-glibc-2.35/lib/libc.so.6(+0x3d2a0) [0x7f5fe94412a0]
#1  0x7f5ec3ba3359 in TechDrawGui::ViewProviderViewPart::fixSceneDependencies() from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/TechDrawGui.so+0x19
#2  0x7f5ec3b8c486 in TechDrawGui::ViewProviderPage::fixSceneDependencies() from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/TechDrawGui.so+0x96
#3  0x7f5ec3af631a in TechDrawGui::MDIViewPage::onMsg(char const*, char const**) from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/TechDrawGui.so+0xea
#4  0x7f5fed34dfeb in Gui::Application::sendMsgToActiveView(char const*, char const**) from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x3b
#5  0x7f5fed46d77e in Gui::Command::_invoke(int, bool) from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x1fe
#6  0x7f5fed46db48 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x108
#7  /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5(+0x2deecd) [0x7f5fe9cdeecd]
#8  0x7f5fea974ad2 in QAction::triggered(bool) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Widgets.so.5+0x32
#9  0x7f5fea977484 in QAction::activate(QAction::ActionEvent) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Widgets.so.5+0xa4
#10  0x7f5fed4d49bf in Gui::ShortcutManager::onTimer() from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0xcf
#11  0x7f5fed4d5a2a in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x52a
#12  0x7f5fed4d6586 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x216
#13  0x7f5fe9caa3de in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5+0x7e
#14  0x7f5fea97ab28 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Widgets.so.5+0xb8
#15  0x7f5fed4229c8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x68
#16  0x7f5fe9caa77a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5+0x11a
#17  0x7f5fea17daec in QShortcutMap::dispatchEvent(QKeyEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Gui.so.5+0x42c
#18  0x7f5fea17e46b in QShortcutMap::tryShortcut(QKeyEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Gui.so.5+0x5b
#19  0x7f5fea12b849 in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Gui.so.5+0x179
#20  0x7f5fea14d05d in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Gui.so.5+0x7d
#21  0x7f5fea12859c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Gui.so.5+0xac
#22  /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5XcbQpa.so.5(+0x683ba) [0x7f5fe57233ba]
#23  /gnu/store/nb40pwd37v6i1g4b1fq4l6q4h9px3asr-glib-2.72.3/lib/libglib-2.0.so.0(g_main_context_dispatch+0x24b) [0x7f5fe6d2498b]
#24  /gnu/store/nb40pwd37v6i1g4b1fq4l6q4h9px3asr-glib-2.72.3/lib/libglib-2.0.so.0(+0x55bf8) [0x7f5fe6d24bf8]
#25  /gnu/store/nb40pwd37v6i1g4b1fq4l6q4h9px3asr-glib-2.72.3/lib/libglib-2.0.so.0(g_main_context_iteration+0x2f) [0x7f5fe6d24c9f]
#26  0x7f5fe9cfe804 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5+0x64
#27  0x7f5fe9ca92fb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5+0x11b
#28  0x7f5fe9cb0f4c in QCoreApplication::exec() from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5+0x7c
#29  0x7f5fed35e6a8 in Gui::Application::runApplication() from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x2528
#30  /gnu/store/8jixnjrwh4jb9c6ig5ry49g0dkmy0jj5-profile/bin/FreeCAD() [0x408995]
#31  /gnu/store/gsjczqir1wbz8p770zndrpw4rnppmxi3-glibc-2.35/lib/libc.so.6(+0x291f7) [0x7f5fe942d1f7]
#32  /gnu/store/gsjczqir1wbz8p770zndrpw4rnppmxi3-glibc-2.35/lib/libc.so.6(__libc_start_main+0x7c) [0x7f5fe942d2ac]
#33  /gnu/store/8jixnjrwh4jb9c6ig5ry49g0dkmy0jj5-profile/bin/FreeCAD() [0x408c31]
```

I'm not sure it is related to the Ctrl-Z action because it crashed earlier with a similar error message, but I cannot remember what I did:

```
Unhandled std::exception caught in GUIApplication::notify.
[... 18 lines cut ...]
#16  0x7f5b09780b40 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Widgets.so.5+0x1a0
#17  /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Widgets.so.5(+0x1d1d78) [0x7f5b097d1d78]
#18  /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Widgets.so.5(+0x1d4d75) [0x7f5b097d4d75]
#19  0x7f5b0977aaef in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Widgets.so.5+0x7f
#20  0x7f5b0c2229c8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x68
#21  0x7f5b08aaa77a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5+0x11a
#22  0x7f5b08f50d07 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Gui.so.5+0x677
#23  0x7f5b08f2859c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Gui.so.5+0xac
#24  /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5XcbQpa.so.5(+0x683ba) [0x7f5b045233ba]
#25  /gnu/store/nb40pwd37v6i1g4b1fq4l6q4h9px3asr-glib-2.72.3/lib/libglib-2.0.so.0(g_main_context_dispatch+0x24b) [0x7f5b05c1f98b]
#26  /gnu/store/nb40pwd37v6i1g4b1fq4l6q4h9px3asr-glib-2.72.3/lib/libglib-2.0.so.0(+0x55bf8) [0x7f5b05c1fbf8]
#27  /gnu/store/nb40pwd37v6i1g4b1fq4l6q4h9px3asr-glib-2.72.3/lib/libglib-2.0.so.0(g_main_context_iteration+0x2f) [0x7f5b05c1fc9f]
#28  0x7f5b08afe804 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5+0x64
#29  0x7f5b08aa92fb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/c0q9vkkgckx751y3hrc3v1c5ywdl37fk-qtbase-5.15.8/lib/libQt5Core.so.5+0x11b
[... cut ...]
#31  0x7f5b0c15e6a8 in Gui::Application::runApplication() from /gnu/store/y2lp2qg6i44d371rsbjzzlr7m7zjf2dw-freecad-0.21.0/lib/libFreeCADGui.so+0x2528
#32  /gnu/store/8jixnjrwh4jb9c6ig5ry49g0dkmy0jj5-profile/bin/FreeCAD() [0x408995]
#33  /gnu/store/gsjczqir1wbz8p770zndrpw4rnppmxi3-glibc-2.35/lib/libc.so.6(+0x291f7) [0x7f5b0822d1f7]
#34  /gnu/store/gsjczqir1wbz8p770zndrpw4rnppmxi3-glibc-2.35/lib/libc.so.6(__libc_start_main+0x7c) [0x7f5b0822d2ac]
#35  /gnu/store/8jixnjrwh4jb9c6ig5ry49g0dkmy0jj5-profile/bin/FreeCAD() [0x408c31]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-08-20:
So the steps to duplicate are:
- create some views,
- delete some views,
- press ctl-z (hoping for undo)
- boom?

luzpaz, 2023-08-21:
CC @rezahousseini

WandererFan, 2023-08-24:
"basic_string::_M_construct null not valid"  generally means that somebody is attempting to create a std::string by passing a null pointer in the constructor. 

But neither of TechDrawGui::ViewProviderViewPart::fixSceneDependencies() or  TechDrawGui::TaskDimension::reject() create std::strings.

WandererFan, 2023-08-24:
I tried unsuccessfully to duplicate this by deleting Views and pressing Ctl-Z in three different flavours of v0.21 (local build, AppImage and Snap).

We must be missing some information about the configuration or the action sequence.

rezahousseini, 2023-08-25:
I was changing strings in the template, and in the view annotations, maybe this is the source of the error? I'll try to reproduce the issue, as it seems the backtrace is not sufficient.

luzpaz, 2023-09-09:
@rezahousseini any progress ?

rezahousseini, 2023-09-12:
I could not reproduce the segfault. I think if the backtrace does not pinpoint to a problem then we can close the issue..

luzpaz, 2023-09-12:
Thanks. 

Hii @wwmayer if you can't make heads or tails of this backtrace, would you kindly close this ticket ?

wwmayer, 2023-09-12:
The callstack at least is correct but I couldn't see either where a null pointer is assigned to a string.

As a test I added this line to ViewProviderViewPart::fixSceneDependencies(): `std::string text = nullptr;` and when following the procedure I got the error `The error message is: basic_string::_M_construct null not valid` 
But FreeCAD didn't crash.

closed 2023-09-13 by WandererFan

andre-caldas, 2023-09-24:
I believe the method called by "notify" is at some point calling Document::getNameInDocument(). However, because the node has already been removed from the document (ctrl-z), its pcNameInDocument is set to nullptr, to indicate it does not belong to the document pointed by "_pDoc".

Unfortunately, as things are now, you cannot call getNameInDocument() and assume the returned value is a valid string. Usually it is, but when objects that are removed from the document are put in a Transaction, the pointer is made null. And this null pointer is used, for example, to know if the Transaction owns the object (needs to delete on destruction) or not.

If you check the returned value you can avoid this problem. It would be nicer if getNameInDocument() always returned a valid string... but I don't know how many people depend on the current behaviour: returning nullptr.

luzpaz, 2023-09-24:
CC @WandererFan :point_up_2:

andre-caldas, 2023-10-08:
It seems to me that we use _pDoc even when the object does not belong to the document anymore. It seems that setting _pDoc to nullptr is not an option.

Maybe, instead of setting DocumentObject::pcNameInDocument to nullptr, we could simply give it a sensible default. Something like "??DoesNotBelongToADocument??".

And also, we could have a "bool DocumentObject::_isAttachedToDocument" and use it instead of comparing pcNameInDocument to nullptr.

I am assuming:
1. No one uses getNameInDocument() to check if object is attached to a document. (we have to search and change this to a call to isAttachedToDocument()).
2. No one assumes that a not nullptr getNameInDocument() implies that `_pDoc != nullptr`. (we have to search and add a check to see if isAttachedToDocument()).
3. No one assumes that when not nullptr, the value returned by getNameInDocument() will exist in _pDoc->objectMap. (here, we also need to check if isAttachedToDocument() is true).

andre-caldas, 2023-10-08:
I am tying to make `DocumentObject::getNameInDocument()` return an std::string.

I found it unbelievable... but there is a `DocumentObject::_outListMap` that is a map which key is actually the address of the internal `const char*` of the `std::string` key for `DocumentP::objectMap`.

Now, that is a big twist to me... I feel a bit dizy... and a little scared... :-(

I guess we could have two methods:
1. `std::string getNameInDocument() const`
2. `const char* getDagKey() const`

I will try...